moggieuk/Happy-Hare #555: Error with mmu_utils.py after update

raetherx, 2024-12-14:
I am receiving the following error on boot after updating to the latest happy hare, klipper, etc.  I did run the full klipper update, happy hare install.sh, happy hare klipper screen update, and so on but I am still getting this error.  Attached is the klippy log from my system.

Klipper reports: ERROR

Internal error during connect: invalid syntax (mmu_utils.py, line 92)

Once the underlying issue is corrected, use the "RESTART"
command to reload the config and restart the host software.
Printer is halted

[klippy (4)(1).log](https://github.com/user-attachments/files/18134201/klippy.4.1.log)

moggieuk, 2024-12-14:
looking...

moggieuk, 2024-12-14:
Ah, I see you are using Python 2.  I am suppose to be supporting that so will fix but you really should upgrade to python 3 soon.  Python 2 has been a dead language since Jan 1st 2020!

raetherx, 2024-12-14:
😱 I thought I had upgraded to python 3 already… I’ll fix that. Thank you!

moggieuk, 2024-12-14:
Before you do... I just pushed a fix. I can't test on python 2 but it is an obvious incompatibility.

Just force `update-manager` to see the new version...

raetherx, 2024-12-14:
I ran the ~/Happy-Hare/install.sh and it came back online immediately. Tomorrow I'll reinstall Klipper to migrate to python 3 to correct my mistake, but thank you for pushing the fix.

closed 2024-12-14 by raetherx