moggieuk/Happy-Hare #557: LEDs fail startup check

iamnotahippy, 2024-12-14:
Since the new LED code changes, the LEDs in my ERCF fail startup checks, although I haven't changed my config for 6 months. My ERCF only has a chain of 8 neopixels at the exits. 

Config and logs attached. 
[mmu_hardware.cfg.txt](https://github.com/user-attachments/files/18134315/mmu_hardware.cfg.txt)
[mmu.log.txt](https://github.com/user-attachments/files/18134317/mmu.log.txt)
[klippy.log (2).txt](https://github.com/user-attachments/files/18134318/klippy.log.2.txt)

moggieuk, 2024-12-14:
You are definitely on the "main" branch? 

The setup does look correct.  I see you trying different led chain lengths...  I need to review the code a little to see if I can guess what might be happening.

BTW: I see these errors:
```
MMU: Improper setup: Filament sensor toolhead is not defined in [mmu_sensors]
MMU: Improper setup: Filament sensor extruder is not defined in [mmu_sensors]
```
This means that you are defining toolhead and extruder sensors by hand and instead should change to defining then in the `[mmu_sensors]` section

moggieuk, 2024-12-14:
Also, what version of klipper are you running?

iamnotahippy, 2024-12-14:
I just pulled the main branch after you fixed the toolchange bug earlier today. 

And I don't understand what you mean about the toolhead and extruder sensors. I have them configured under [mmu_sensors] in mmu_hardware.cfg

```extruder_switch_pin: ^EBBCan:PB8
toolhead_switch_pin: ^EBBCan:PB9```

iamnotahippy, 2024-12-14:
I'm running Kalico (formerly DangerKlipper) v0.12.0-551-gb500c382

moggieuk, 2024-12-14:
Well, that could be the issue... ?  If you switch to regular klipper does it work?

moggieuk, 2024-12-14:
That would explain the Happy Hare `TRSYNC` error as well.

iamnotahippy, 2024-12-14:
So much of my config relies on Kalico features, I couldn't feasibly switch between them to test. I can comment out the TRSYNC setting in my Kalico config, but it's been like that for quite some time without issue.

moggieuk, 2024-12-14:
The problem is that I don't support Danger Klipper.  I've tried to keep it working but I can't guarantee..

iamnotahippy, 2024-12-14:
I haven't been on regular Klipper for 6 months at least.

iamnotahippy, 2024-12-14:
If I'm the only person affected, I'm happy to fork HH and revert to the old LED code rather than waste your time.

moggieuk, 2024-12-14:
Well, it's not just LEDs...:
(i) I'm seeing incompat with TRSYNC ... you can use the danger klipper version for that and turn setting off in HH.
(ii) The extruder and toolhead sensor error is concerning because I see problems with runout and false triggers. I know this works with regular klipper.
(iii) LED startup. I've reviewed the diff and the changes are very small .. yet an extra validation check.

iamnotahippy, 2024-12-14:
(i) I've made the change. Thanks!
(ii) those are primarily hardware issues. My extruder sensor is mounted above my gears, which causes me to have to engage in some kludging to get it to work well. My pregate sensors also needed a bit of adjustment recently, which probably caused the runouts you see. 
(iii) ? I'll just fork and revert the changes. No big deal.

moggieuk, 2024-12-14:
On the danger-klipper version.   I just switched to test - leds work and I don't see the sensor error.

I'm on the mainline danger-klipper.  Are you on one of their feature branches?

iamnotahippy, 2024-12-14:
DangerKlipper has recently changed their name and repo. They're now called Kalico, and can be found at 

https://github.com/KalicoCrew/kalico

Current version from the main branch is v0.12.0-551-gb500c382

iamnotahippy, 2024-12-14:
You're right that there's a connection. I rolled back Kalico to commit e0901b2 and the LED function has returned. Commit 136149b inadvertently broke some other things, so I wouldn't be surprised if the problem is there. I'll bring it up to the Kalico team.

Their discord is over at https://kalico.gg/discord if you're interested. 

Again, thanks so much for all the time and effort you've put into this!

moggieuk, 2024-12-14:
Well, it wasn't the way I wanted to spend today but I'm finding many issues with the latest danger-klipper.   One is a fundamental change to the way they run kipper meaning that all python class "import" statements are not the incorrect path.  That means that any "isinstance()" check fails.

That breaks, many things in Happy Hare!

moggieuk, 2024-12-14:
Ok.  I have updated Happy Hare to work with the latest Danger-Klipper.

 I can see that the Kalico team has actually cleaned up some holes in klipper but this may make other plugins have problems too.

closed 2024-12-14 by moggieuk

kageurufu, 2024-12-14:
We had tested things (I've been using that code for months) but it seems it wasn't perfect. I'm gonna look at what you had to change and see if I can fix the compatibility hooks