moggieuk/Happy-Hare #558: Spoolman doesn’t count filament use correctly

igiannakas, 2024-12-15:
When doing multi color prints, I’ve noticed that spoolman doesn’t account for the full material consumption. For example the print below was about 30grams off, mostly on the white color (as this was the one consumed the most).

[image]

screenshot from mainsail - notice the discrepancy between the slicer estimate and the actual consumption as measured by klipper:

[image]

I weighed the purged material, model and prime tower and the slicer is within a few grams of the actual print - see also here for testing on the slicer side: https://github.com/SoftFever/OrcaSlicer/pull/7508

basically what I’m seeing is spoolman recording less filament consumed, which can be a bit of an issue as I need to periodically re weigh the spools and adjust the material remaining manually.

for example, the below white spool should have had around 90 grams remaining according to spoolman yet it was under 20 grams left (it was on the last set of loops around the Center of the spool)
[image]

please note, this only happens with multi color prints. Single color prints work correctly and spoolman reconciles down to the gram between slicer, actual and spool weight.

Observations when printing a multi color model:
1. During printing the model, spoolman deducts the weight correctly
2. During purging with the blobifier I do notice filament consumed, so that also is being measured

That leads me to believe that somewhere in the retractions/ cut/ unload / reload logic spoolman doesn’t potentially count the filament remaining in the hotend correctly (?) hence the discrepancy which grows the more tool swaps you do. 

This is probably expected as the extruder retracts more than the filament does when pulling the filament out from the toolhead, because it carries on retracting when the filament has already cleared the extruder top by design to ensure the filament is free from the extruder before running the Bowden unload move. 

This is probably being counted as extra filament being re wound but obviously it’s not there as the filament has already cleared the extruder. Specifically this variable I’m suspecting: 
toolhead_unload_safety_margin: 15

15mm over a few hundred swaps accumulates.

Or maybe the toolhead residual filament plus the remaining fragment after the cut is not counted?

Maybe a better strategy would be to re consider at what points the filament is counted as loaded in the extruder? Or increment the filament consumption by the remaining fragment after cut plus the toolhead residual filament value?

In general going conservative and having more material left over would be better than having less on the spool than what spoolman reports.

moggieuk, 2024-12-20:
I love your attention to detail.   I'll need to review the code logic to see exactly where the spool id is switched... it make omit purging for example...

igiannakas, 2024-12-20:
I think purge in the blobifier is counted correctly - else the variance also would have been significantly more. My instinct is telling me it’s a small error that accumulates over many tool changes but I may be wrong too 😂

Mordenkainen, 2025-01-29:
Based on watching the SpoolMan UI as well as Mainsail UI in parallel, I am NOT seeing the filament purged by Blobifier being accounted for.

When a swap is happening, the tool is unloaded, and Spoolman dismounts the spool, then the new filament is loaded and the toolhead is purged. At this point the Mainsail UI does not show any spool mounted, and SpoolMan does not show any filament usage for the purge. Once the purge is complete, the Mainsail UI shows the spool mounted, and Sppolman starts showing filament usage.

I have not taken a peek at the code yet, but I assume that the spool does not get registered with Spoolman until the swap is complete, including any user specified macros like the Blobifier purge.

igiannakas, 2025-02-03:
Done a bit more investigation here:
https://github.com/SoftFever/OrcaSlicer/pull/7508

Looks like the estimates in Klipper - slicer - actual purge are consistent with each other. It's only the spoolman counting that is off.

igiannakas, 2025-02-03:
> Based on watching the SpoolMan UI as well as Mainsail UI in parallel, I am NOT seeing the filament purged by Blobifier being accounted for.
> 
> When a swap is happening, the tool is unloaded, and Spoolman dismounts the spool, then the new filament is loaded and the toolhead is purged. At this point the Mainsail UI does not show any spool mounted, and SpoolMan does not show any filament usage for the purge. Once the purge is complete, the Mainsail UI shows the spool mounted, and Spoolman starts showing filament usage.
> 
> I have not taken a peek at the code yet, but I assume that the spool does not get registered with Spoolman until the swap is complete, including any user specified macros like the Blobifier purge.

That sounds like a very plausible cause!!

igiannakas, 2025-02-03:
@moggieuk 

Maybe the fix here is to move the spoolman spool selected notification below above this segment here:

`self._spoolman_activate_spool(self.gate_spool_id[gate]) # Activate in Spoolman`

```
if macros_and_track:
                with self._wrap_track_time('post_load'):
                    # Restore the expected sync state now before running this macro
                    sync = (self.is_printing() and self.sync_to_extruder) or self._standalone_sync
                    self.sync_gear_to_extruder(sync, grip=True, current=True)
                    if self.has_blobifier:
                        with self.wrap_action(self.ACTION_PURGING):
                            self.wrap_gcode_command(self.post_load_macro, exception=True, wait=True)
                    else:
                        self.wrap_gcode_command(self.post_load_macro, exception=True, wait=True)
```
Currently it notifies spoolman after the load sequence is completed?

igiannakas, 2025-02-04:
> Based on watching the SpoolMan UI as well as Mainsail UI in parallel, I am NOT seeing the filament purged by Blobifier being accounted for.
>
> When a swap is happening, the tool is unloaded, and [...]

Thank you for this - indeed that was the cause. I’ve raised a PR that addresses this :) testing with a multi color print today but so far so good!

closed 2025-02-04 by moggieuk

AdamDrefs, 2026-03-01:
Kinda late to be asking, but does this include the built-in purge macro? I had my custom purge macro defined in place of the typical MMU_PURGE or BLOBIFIER macros, as such
```
force_purge_standalone: 1	
purge_macro: 'PURGE' #custom purge macro
extruder_purge_current: 100		# % of extruder current (100%-150%) to use when purging (100 to disable)
```
However when I finally got around to adding Spoolman, it wasn't tracking my purged material. I have moved my macro to the post-load macros, which looking at the PR will definitely work, but I still think that the PURGE macros should be included as well.

AdamDrefs, 2026-03-01:
One thing to note is that Klipper didn't track my purge filament either for some reason, so it's not a Happy-Hare or Spoolman specific error

AdamDrefs, 2026-03-03:
Turns out I was mistaken this whole time. Spoolman had correctly tracked my filament usage, I just had made a few mistakes in the initial setup and some math stuff.

PR #619 (2025-02-04): Fix current spool in spoolman being activated after the load sequence is complete
Fixes active spool in spoolman being set after the load sequence is complete. 

The current behaviour triggers the post load macros and then associates the spool from spoolman, meaning that any blobifier purging is not counted against the active filament.

Have moved both the load and unload association from Spoolman for consistency in the code.

Fixes #558

[image]