moggieuk/Happy-Hare #560: Error importing file_manager during tests.

rnlahaye, 2024-12-16:
When trying to run the unit tests (via `test/runner.sh`), I get the following error:

```
Traceback (most recent call last):
  File "/home/rynildredeth/Happy-Hare/test/components/test_mmu_server.py", line 13, in setUp
    self.subject = MmuServer(MagicMock())
  File "/home/rynildredeth/Happy-Hare/components/mmu_server.py", line 80, in __init__
    self.setup_placeholder_processor(config)
  File "/home/rynildredeth/Happy-Hare/components/mmu_server.py", line 736, in setup_placeholder_processor
    from .file_manager import file_manager
ModuleNotFoundError: No module named 'components.file_manager'
```

This error occurs both on my laptop and when ssh-ed into the printer.

moggieuk, 2024-12-21:
I'm away from my printer / system, but I believe you need to `cd` into the `components/file_manager` or at least the `components` directory under moonraker for this to work because that is where `mmu_server.py` will normally be loaded by moonraker.

moggieuk, 2026-10-21:
I'm closing this as part of a tidy-up of old v3 issues. Happy Hare v4 is a massive architectural change from v3, so it isn't possible to address these on v3 any more.

The old test runner is gone in v4; tests now run with "make test", which sets up its own environment.

If this still exists in Happy Hare v4 (v4.1 is the current release), **please resubmit** it as a new issue (with your `mmu.log` if relevant) and it will be reviewed.

closed 2026-10-21 by moggieuk