moggieuk/Happy-Hare #561: Fix MMU_CALC_PURGE_VOLUMES

jacksky6, 2024-12-17:
I am very happy that 3.0 has this feature.
But it seems that there is something wrong with the result of `MMU_CALC_PURGE_VOLUMES` calculation. It seems to be reversed.

For example, my T0 is black and T1 is white. After I execute `MMU_CALC_PURGE_VOLUMES`, 
I use `MMU_SLICER_TOOL_MAP PURGE_MAP=1` to check the result as follows:
```
$ MMU_SLICER_TOOL_MAP PURGE_MAP=1
// No slicer tool map loaded
// Purge Volume Map (mm^3):
// To -> T0   T1   T2   T3
// T0    -    79   -    -
// T1   559   -   559  559
// T2    -    79   -    -
// T3    -    79   -    -
```

Switching from black T0 to white T1 flushes out 79; 
switching from white T1 to black T0 flushes out 559.
This is obviously the opposite.

The result should be:
```
// Purge Volume Map (mm^3):
// To -> T0      T1
// T0    -       559   
// T1   79        -
```

See if this is the case.

moggieuk, 2024-12-20:
Thanks for the feedback.  I have to admit that I verified the code worked but didn't scrutinize the actual volumes. I will review as soon as I get a free moment..

moggieuk, 2024-12-21:
@jacksky6  I'm travelling over the holidays and it is difficult to verify a fix. However if you can make this change and verify I can push a patch.

On line 1118 in `mmu.py` (you can find in `Happy-Hare/extras/mmu/mmu.py`, modify:
```python
        # Build purge volume map (x=to_tool, y=from_tool)
        should_calc = lambda x,y: x < len(tool_colors) and y < len(tool_colors) and x != y
        purge_volumes = [
            [
                purge_vol_calc.calc_purge_vol_by_hex(tool_colors[y], tool_colors[x]) if should_calc(x,y) else 0
                for y in range(self.num_gates)
            ]
            for x in range(self.num_gates)
        ]
        return purge_volumes
    ```
    to:
    ```python
        # Build purge volume map (x=to_tool, y=from_tool)
        should_calc = lambda x,y: x < len(tool_colors) and y < len(tool_colors) and x != y
        purge_volumes = [
            [
                purge_vol_calc.calc_purge_vol_by_hex(tool_colors[y], tool_colors[x]) if should_calc(x,y) else 0
                for x in range(self.num_gates)
            ]
            for y in range(self.num_gates)
        ]
        return purge_volumes
    ```
    Note the ONLY change is the `x` -> `y` and `y` -> `x` on two of the lines

If you can make this change and let me know if that correctly reverses the generated map...

ningpj, 2024-12-21:
I made the change.  HH calc's seem sensible now.  Screensnap with Before / After patch & using ``mmu_calc_purge_volumes`` to update.  Included snap of Orca volumes and gate mapping.  Blacks in gates 0 & 7 in my case. 
[image]

moggieuk, 2024-12-22:
@ningpj thank you so much.   I will commit with confidence (even though I'm not by my printer)..

moggieuk, 2024-12-22:
Patch pushed.  Simple quick `update-manager` update will fix. @jacksky6

closed 2024-12-22 by moggieuk