moggieuk/Happy-Hare #562: Issue installing Happy Hare on K1 Max.

pvilbig, 2024-12-18:
So I am trying to install HH on my K1 Max but I don't understand how to upgrade the firmware no the Max as per the quick install option for 3ms. and hep would be greatly appreciated!

moggieuk, 2024-12-21:
I don't have a K1 so an can't help without logs or specific errors.  You could ask in the 3ms forum on the Happy Hare discord or attach screen capture / logs of what is failing...

pvilbig, 2024-12-21:
So here is what comes up 

[image]

moggieuk, 2024-12-22:
Ok, that isn't anything to do with Happy Hare running on Klipper.  That is from Klipperscreen.

That error indicates that you are not running the HH fork of kliperscreen because the base version can't handle parameters that are lists.

Try these steps to verify / fix

What does this report?
```
cd KlipperScreen
git remote -v
```

It should report something like:
```
origin    git@github.com:moggieuk/KlipperScreen-Happy-Hare-Edition.git (fetch)
origin    git@github.com:moggieuk/KlipperScreen-Happy-Hare-Edition.git (push)
```
If it does, then try:
```
git pull
```
Finally, run this one more time.
```
cd ~/KlipperScreen/happy_hare
./install_ks.sh -g <number_of_gates>
```
(you actually put the number of gate you want to configure KlipperScreen to handle after `-g`.  E.g. `-g 8` for an 8-gate ERCF.

moggieuk, 2025-01-19:
I'm going to close this Issue because I believe it has been answered.

closed 2025-01-19 by moggieuk