moggieuk/Happy-Hare #563: Disable buffer setting

tkunchick, 2024-12-20:
For setups using rewinders and always pulling from the spool, it may be helpful to be able to disable all buffer settings.

tkunchick, 2024-12-24:
was thinking, adding a buffer length variable might be an easy solution. much like the other length variables.

igiannakas, 2025-01-07:
For reminders it would be interesting to have a per lane speed adjustment - in some cases some of the rewinders can be a bit further away from the printer compared others, necessitating a slower load and unload speed due to varying Bowden lengths from the unit to the rewinder. Now you have to use the lowest value from all to have it be reliable.

moggieuk, 2026-10-21:
Fixed by #1438. The menuconfig "Has filament buffer to catch loose filament?" option now sets `filament_buffer` in `[mmu_unit]`, so you can turn the buffer off. With it off, loads always use `gear_load_speed`, which suits setups with rewinders that always pull from the spool. Before this, the option only showed or hid the buffer speed settings, and ERCF/Tradrack units always behaved as buffered. Adding an eSpooler now turns the buffer off too.

You can also set `filament_buffer: 0` directly in `[mmu_unit]`, and that setting is kept when the config is refreshed.

This will ship in **v4.2.0**. Watch for that release and re-run the installer after upgrading. Leave the option ticked if you do have a buffer.

Per-lane speeds (suggested above) and a per-gate buffer setting (#602) aren't covered here. The setting is still per unit.

moggieuk, 2026-10-21:
#1438 has been merged into `development` and will ship in **v4.2.0**. Once you've upgraded, re-run the installer and untick "Has filament buffer to catch loose filament?" in menuconfig to turn the buffer off. Alternatively, set `filament_buffer: 0` in `[mmu_unit]`.

closed 2026-10-21 by moggieuk