moggieuk/Happy-Hare #564: No filament cutting on first start

emintjeee, 2024-12-20:
Hi,

After upgrading to V3 it stopped with cutting the filament on a first load/unload (if its still loaded).

Regards,
Embit

emintjeee, 2024-12-20:
See at the latest print job (19:36) he try to get the filament of the toolhead without heating/cutting. it fails because the filament is stuck in the print head.
[mmu.log](https://github.com/user-attachments/files/18214164/mmu.log)

3DCoded, 2024-12-20:
I also have this issue.

moggieuk, 2024-12-21:
I looked at the attached `mmu.log`.  Can you confirm that this first unload is the one that should have run the cut-tip macro:
```
14:04:44 Checking all required gates have filament loaded...
14:04:44 > MMU_CHECK_GATE TOOLS=0,1,2,3,4
14:04:44 Unloading current tool prior to checking gates
14:04:44   DEBUG: Unloading tool Unknown
14:04:44 Unloading filament...
14:04:44 [T?] < (*) <<<<<<<<<< (*) << [Ex << (*) << Nz] LOADED 0.0mm
14:04:44   DEBUG: Extracting filament from extruder
14:04:44   DEBUG: Setting servo to down (filament drive) position at angle: 120
14:04:44 > MMU_STATS COUNTER=servo_down INCR=1
14:04:44   DEBUG: Reverse homing up to 117.6mm to extruder sensor (synced) to exit extruder
14:04:49 [T?] < (*) <<<<<<<<<<|(e) .. [Ex .. (t) .. Nz] -76.5mm
14:04:49   DEBUG: Filament should be out of extruder
14:04:49   DEBUG: Unloading bowden tube
14:04:49   DEBUG: Sensors: {'mmu_gate': True}
14:04:55 [T?] < (*) <<........ (e) .. [Ex .. (t) .. Nz] -1097.6mm
14:04:57 [T?] <|(g) .......... (e) .. [Ex .. (t) .. Nz] -1155.5mm
14:04:57 [T?] < (g) .......... (e) .. [Ex .. (t) .. Nz] UNLOADED -1173.0mm
14:04:58 Unload of -1173.0mm filament successful
14:04:58   DEBUG: Autotune: bowden move ratio: 0.000000, Extra homing movement: 0.0mm
14:04:58   DEBUG: Deactivating spool...
```

It does look like there is a bug here....

moggieuk, 2024-12-21:
Ok. I pushed a patch that should address this.  Yes it was a coding error..

closed 2024-12-21 by moggieuk