moggieuk/Happy-Hare #566: No blobifier at RESUME

emintjeee, 2024-12-20:
Hi,

If there is something wrong, and you fix it like load a filement and you do RESUME. its skipped the blobifier. and it directly starts with printing. so than you have still the previous color in the hotend..

Regards
Embit

emintjeee, 2024-12-20:
See file. it happend at 20:08 after RESUME no blobifier.
[mmu (1).log](https://github.com/user-attachments/files/18214454/mmu.1.log)

moggieuk, 2024-12-21:
That is actually as designed. Verifying with @Dendrowen.

If the MMU experiences an error you are expected to load the required filament (including purging) before resuming.  You may be able to call the blobifier macro directly before resuming...

Dendrowen, 2024-12-21:
Correct. It's a post load script so it should purge after loading. If you want to have it executed at print resume you might want to check a resume extension in HH (if there is one) or add BLOBIFIER to your resume macro.

emintjeee, 2024-12-22:
If i add the BLOBIFIER in the resume macro it will disappear after a update. a resume extension is great. are this a option to implement?

closed 2026-10-13 by github-actions[bot]