moggieuk/Happy-Hare #567: Motor sync issue during tip forming.

AdamDrefs, 2024-12-21:
So I've been having issues with tip forming during prints, where the tip forms perfectly outside a printing scenario, but causes blobs and ejection issues during prints. I have now found out that the issue seems to be caused by a motor sync issue. I had motor sync enabled, but I found that during tip forming, the MMU gear motor wouldn't sync with the extruder, despite the servo being in the down position. This meant that my extruder motor was literally fighting against the MMU, causing it to not be able to perform it's movements and as such cause blobbing and tip forming issues. I am on happy hare v3.0.1-49-g9d06d262. If needed, I can provide any log files. For now however, I have just disabled sync.

moggieuk, 2024-12-21:
Please provide `mmu.log`.  Note that syncing during standalone tip-forming is controlled separately from in print but the servo should be in the correct position based on this setting.

AdamDrefs, 2024-12-21:
Heres my mmu.log
[mmu.log](https://github.com/user-attachments/files/18219483/mmu.log)
and klippy.log too
[klippy (1).log](https://github.com/user-attachments/files/18219484/klippy.1.log)

ningpj, 2024-12-21:
When you had sync to extruder enabled did you ever set ``sync_form_tip = 1`` as well? From ``klippy.log`` it looks like it was never set and could be a contributing factor as HH "should" have raised the servo when ``sync_form_tip = 0``.
Can you enable both sync options to see if it works around the issue

Explorist210, 2024-12-21:
Ive been having the exact same issue. However, when I enable sync_form_tip it works fine although its been causing timer too close issues. Out of print the servo would lift to allow tip forming, in print the servo remains engaged and causes run out issues.

AdamDrefs, 2024-12-22:
I've never even seen the option for "sync_form_tip", but I'll try to find it and see if it works.

AdamDrefs, 2024-12-22:
Ah, so it seems I overlooked "sync_form_tip" and had it set to 0. However, this still doesn't explain why the servo would be in the wrong position during tip forming. I totally agree that it should have been in the up position during tip forming because the sync for tip forming was disabled, however it was not in the up position, and i know 100% that sync was the issue since it was resolved as soon as I disabled sync. I don't know if this is an issue with v3.0 b/c I just upgraded to 3.0 before I started having issues, so that could be the culprit. However, it could simply be some error in judgement on my behalf.

ningpj, 2024-12-22:
i expect its a bug that needs to be looked at as HH should have either released and raised the servo when trying to form the tip with ``sync_form_tip=0`` OR set it and formed the tip with sync enabled.

Explorist210, 2024-12-22:
My thoughts Too.

closed 2025-02-04 by github-actions[bot]