moggieuk/Happy-Hare #568: Enabled steppers on default since happy-hare v3.0.x

Niran078, 2024-12-21:
Seems like since happy hare v3.0.x steppers are turned on by default on startup. This is a problem since I run my raspberry pi with klipper 24/7. While printing my 24v PSU is turned on to provide power to the the printer and my mmu mcu. 

However since the new happy hare update I cannot fully boot into mainsail and klipper because it shutsdown because of a stepper error on bootup, and since I can only turn on my 24v PSU through mainsail I cannot easily fix this error without physicaly switching the 24v PSU relay on everythime I restart klipper. 

I already turned off **variable_home_mmu** & **variable_auto_home** but it still looks like happy-hare enables the selector and gear steppers on startup. Once klipper goes into idle timout state  (after 10 minutes of doing nothing) my 24v PSU turns off and klipper just keeps running without a problem (seems like happy-hare disables the steppers just fine after idle timout) 

FIX: I would like an option to have the gear and selector steppers disabled on start-up

moggieuk, 2024-12-21:
I think this is the "state persistence" in Happy Hare.  The default is now to always restore state (which if you are using ERCF or Tradrack means setting the selector stepper position and setting the correct servo position).  Is guess there could be a "bypass startup" but I'd have to think about other implications of this.  There are actually quite a few cases on startup where a motor could be activated.

Niran078, 2024-12-21:
Thank you for the fast reply!

Do you know if maybe there could be a workaround just for my very specific case?

moggieuk, 2024-12-22:
Well, it is all part of the "bootup" code that HH goes through.  The only way I can see to disable at the moment would be to edit your version of `mmu.py` and find this line:
```python
        # Initializer tasks
        self.gcode.register_command('__MMU_BOOTUP', self.cmd_MMU_BOOTUP, desc = self.cmd_MMU_BOOTUP_help) # Bootup tasks
```
Comment out this line and then define your own macro called `__MMU_BOOTUP`.  

This will cause HH to call you macro on startup rather than the built-in one. Beware that a few things won't be done:
1. Display welcome message
2. Correct an incorrect filament location
3. Clear TTG map
4. Force selection of gate
5. Auto home MMU.

This would be untested and it might be ok if you `MMU_HOME` before attempting to print and also manually clear TTG map if necessary..

closed 2026-11-03 by moggieuk