moggieuk/Happy-Hare #569: No Module named 'extras.mmu'

tiaanv, 2024-12-21:
Version:
happy-hare
v3.0.1-49-g9d06d262

Updated to latest build, now I get an error:
[image]

From klippy log:
```
Unhandled exception during connect
Traceback (most recent call last):
  File "/home/pi/klipper/klippy/klippy.py", line 281, in _connect
    self._read_config()
  File "/home/pi/klipper/klippy/klippy.py", line 214, in _read_config
    self.load_object(config, section_config.get_name(), None)
  File "/home/pi/klipper/klippy/klippy.py", line 198, in load_object
    self.objects[section] = init_func(config.getsection(section))
  File "/home/pi/klipper/klippy/extras/mmu_sensors.py", line 247, in load_config
    return MmuSensors(config)
  File "/home/pi/klipper/klippy/extras/mmu_sensors.py", line 146, in __init__
    from extras.mmu import Mmu # For sensor names
ModuleNotFoundError: No module named 'extras.mmu'
```
seems to originate from the file mmu_sensors.py:
```
class MmuSensors:

    def __init__(self, config):
        from extras.mmu import Mmu # For sensor names

        self.INSERT_GCODE = "__MMU_SENSOR_INSERT"
        self.REMOVE_GCODE = "__MMU_SENSOR_REMOVE"
        self.RUNOUT_GCODE = "__MMU_SENSOR_RUNOUT"

        self.printer = config.get_printer()
        event_delay = config.get('event_delay', 0.5)

```

any help would be appreciated.  I tried a couple of variations for the import, but I'm not much of a python dev,

emintjeee, 2024-12-21:
I have this too, comment for follow :)

moggieuk, 2024-12-21:
https://github.com/moggieuk/Happy-Hare/wiki/Troubleshooting-and-Common-Issues#-no-extrasmmu-error

The discord forums would have helped you out instantly..

tiaanv, 2024-12-21:
I am so sorry. I should have done more homework, but a Google search turned up nothing, and I don't Discord enough. There is nothing worse than having to spoonfeed dummies like me.  Thx for the pointer.

closed 2024-12-21 by tiaanv