moggieuk/Happy-Hare #573: Voron IDEX with ERCF?

mg197879, 2024-12-23:
I am in the process of completing an AWD Voron IDEX. I currently have one 8 channel ERCF attached to the left extruder (T0). The second extruder (T1) is where the opportunities start. The ERCF from what i can tell only starts at T0. Is there any way to renumber the first gate from T0 to T2? Or if anyone knows how to change the IDEX configuration for the second tool head from T1 to T8?

moggieuk, 2025-01-06:
Hi!    You are on the "bleeding edge" with a setup like this.  Happy Hare internally is always 0-based.  It would be possible to adjust the `Tx` macros that are created in `mmu_macro_vars.cfg` to map say T2 command to `MMU_CHANGE_TOOL TOOL=0`.  That may work but I'm not sure what other issues it may lead to.

The other untested (my me) setup is with IDEX printers and configuring one extruder to work with the MMU.  I coded this to work but I've never had the opportunity to test in practice.

mg197879, 2025-01-06:
Ive been able to get it to work with the klipper toolchanger. It works like this (so far). Basically if I call T0 it selects the left toolhead and does a toolchange on ERCF0 to gate 0, T1 selects the right toolhead and toolchange to gate 0 on ERCF1. T2 selects Left toolhead and gate 1 on ercf and so on. Still trying to get all the bugs worked out. My Frankenstein printer is AWD IDEX and 600 cubed with dual ERCF lol. First build was a little ambitious.

[Use caution with links & attachments]

moggieuk, 2026-10-21:
I'm closing this as part of a tidy-up of old v3 issues. Happy Hare v4 is a massive architectural change from v3, so it isn't possible to address these on v3 any more.

In v4 each MMU unit can be attached to its own extruder (extruder_name in the unit config), which is worth a look for a two-ERCF IDEX build.

If this still exists in Happy Hare v4 (v4.1 is the current release), **please resubmit** it as a new issue (with your `mmu.log` if relevant) and it will be reviewed.

closed 2026-10-21 by moggieuk