moggieuk/Happy-Hare #576: Software only configuration for blobifier

appleimperio, 2024-12-26:
Can you add an option to use the software only configuration to Blobifier to a printer like a Switchwire. Thanks

jack-samo10000, 2024-12-26:
Hello, we tried to solve the issue.

This is what we did:

Added a new variable 'variable_software_only' to enable software-only mode. Modified relevant sections to skip hardware-dependent operations when in software-only mode. Updated initialization to check for software-only mode and adjust behavior accordingly.

You can review changes in this commit: https://github.com/jack-samo10000/moggieuk-Happy-Hare-576/commit/4c1f6acf363aa57b0da0e811fae3eba642fcbfc5.

> [!CAUTION]
> Disclaimer: The concept of solution was created by AI and you should never copy paste this code before you check the correctness of generated code. Solution might not be complete, you should use this code as an inspiration only.

---

Latta AI seeks to solve problems in open source projects as part of its mission to support developers around the world. Learn more about our mission at https://latta.ai/ourmission . If you no longer want Latta AI to attempt solving issues on your repository, you can block this account.

moggieuk, 2025-01-06:
Blobifier is integrated into Happy Hare but you are welcome to pull out the macro and modify to work standalone..

closed 2026-10-13 by github-actions[bot]