moggieuk/Happy-Hare #577: Variable not matching my printerconfig

lorangthierry, 2024-12-28:
Hi,
I have to replace in mmu_form_tip.cfg the line
    {% set orig_fan_speed = printer.fan.speed %}
with
    {% set orig_fan_speed = printer['output_pin fan0'].value %}

If not I have a firmware crash when I change the temperature for tip forming.
For an update I have to reset the Git Repo and after the Update I have to redo the changes. 
Can I define the printer.fan.speed somewhere else so that I have not to change the read-only file?

ningpj, 2024-12-28:
you can create a copy, rename and modify the tip forming macro to suit.
Set ``form_tip_macro`` in ``mmu_parameters.cfg`` to the new name of your tip forming macro.

lorangthierry, 2024-12-30:
> you can create a copy, rename and modify the tip forming macro to suit. Set `form_tip_macro` in `mmu_parameters.cfg` to the new name of your tip forming macro.

Is also a possibility. I think it is easier to change the file after the update. So I don't have to check the original and my version to differences.

lorangthierry, 2025-01-07:
[printer.cfg.txt](https://github.com/user-attachments/files/18338502/printer.cfg.txt)
Here my printer.cfg. 
It is a stock Qidi X-Plus3 with Klipper v0.12.0-404-g80d185c9
Maybe the fans are wrong defined in the printer.cfg. 
The cfg was downloaded from https://github.com/leadustin

closed 2025-02-21 by github-actions[bot]