moggieuk/Happy-Hare #578: BTT MMB wrong pin

chadakenn, 2024-12-30:
MMU_GEAR_ENABLE=PA8 should be PB8 for v1.1

ningpj, 2024-12-30:
It already sets this (and ``MMU_PRE_GATE_1=PA8``) for BTT MMU 1.1 boards.
Did you select the correct board version when running ``./install.sh`` as there are two options
```
Select mcu board type used to control MMU
1) BTT MMB v1.0 (with CANbus)
2) BTT MMB v1.1 (with CANbus)
3) Fysetc Burrows ERB v1
4) Fysetc Burrows ERB v2
```

chadakenn, 2024-12-30:
> It already sets this (and `MMU_PRE_GATE_1=PA8`) for BTT MMU 1.1 boards. Did you select the correct board version when running `./install.sh` as there are two options
>
> ```
> Select mcu board type used [...]

i did it was all working and then it just stopped one day and i tracked down that pin was changed idk if it was after a update or what

ningpj, 2024-12-30:
i just ran through a clean install and upgrade cycle and the pins were correctly set and unaltered post upgrade. If ``MMU_PRE_GATE_1=PB8`` was also set in ``mmu.cfg`` alias section its likely the interactive install option  ``-i`` was triggered at some stage

moggieuk, 2025-01-06:
@chadakenn not sure what happened in your case (maybe set with an old release) but you can see the pin defs in the file `Happy-Hare/pin_defs` to confirm.  Here is the MMB line:
```
PIN[MMB10,gear_enable_pin]="PA8";                          PIN[MMB11,gear_enable_pin]="PB8";
```
So looks good now for new users.

vajonam, 2025-01-08:
@moggieuk with a new install for BTT MMB 1.1. 

This was left in `  MMU_POST_GEAR_11={gear_sensor_11_pin},` in the mmu.cfg which for whatever reason didn't get replaced

running `Already the latest version: v3.0.1-63-g93d4fda`

vajonam, 2025-01-08:
Copy paste issue https://github.com/moggieuk/Happy-Hare/blob/93d4fda8c22af6ef6e8d67f2ae90c9bf0d90884e/pin_defs#L257 I will open a PR.

moggieuk, 2025-01-08:
Thanks for the PR -- it have been pullled!

closed 2025-01-08 by moggieuk