moggieuk/Happy-Hare #579: Problem beim druckstart mit ercf

Thomsen1982, 2024-12-31:
All necessary gates are loaded. The necessary gates are checked at the start of printing. Right at the first gate, the filament is first retracted, then of course it runs into an error. I thought the error was fixed with the latest version, as it was in the history. But the error still exists.

Greetings 
Thomsen 
[mmu 2.log](https://github.com/user-attachments/files/18280303/mmu.2.log)

moggieuk, 2025-01-06:
I'm not sure but I wonder if this is because of your user of `MMU_EJECT`?

With v3.0.x there is a difference between `MMU_UNLOAD` and `MMU_EJECT`.  The latter will push the filament completely out of the MMU (or at least try to).  `MMU_UNLOAD` will keep it parked in the gate.

More information in the release notes here: https://github.com/moggieuk/Happy-Hare/wiki/Change-Log#v300

Thomsen1982, 2025-01-06:
Yes thanks
That was the Problem

closed 2025-01-06 by Thomsen1982