moggieuk/Happy-Hare #580: Clog triggered by out of sync Synced Extrusion

lettore, 2024-12-31:
Hello and Happy New Year,
I'm experiencing some clog with TPU due to, I guess, out of sync motion by extruder and MMU. While I can guess TPU is not a good candidate for ERCF MMU, I think there can be a solution for this issue.
You already added a sensor logic to manage compression and extension of the filament inside the bowden tube, and while it will be my next project to implement such sensor, for now I have to disable the sync extrusion otherwise it will trigger clog after few meters.
I calibrated the gate with TPU loaded so I'm sure it loads the correct amount of filament, so in config there's little I can do.
What I will suggest for the future is:
1 - have material or gate settings for sync and clog, to explain be able to add modifiers if a specific type of filament is loaded, or as last resort for a specific gate that will be used only with a specific filament.
Something like: Override section
sync_to_extruder-TPU: 0	or sync_to_extruder-GATE5: 0
In this way the software will check if there are specific gate settings, or material settings based on what it's loaded. I think this will be something useful, because it will be possible to change other settings like speed, or default temperature based on specific filament type.
2 - add to the clog macro a option to release the tension before triggering the pause. If the clog happens when sync to extruder is enabled and no tension sensor is installed, it's possible that the filament is present but just held by the MMU gear, so before triggering the fail it should release the servo, let the extruder work alone for a while and see if the filament return moving correctly, if so put down the servo and continue normal operation, otherwise just go forward and pause the print.

moggieuk, 2025-01-06:
Happy New Year.   Definitely some good ideas here.   I was aware of the possible issues with encoder based clog detection and the user of a sync-feedback sensor (at least one that has a movement buffer).  I quite like the per-gate clog detection feature but also want to think about more generic ways to cope with this variability in the bowden length.  It the encoder is after the sync-feedback it wouldn't be an issue. It only occurs if before. (Clog detection is the mismatch of extruder movement to encoder detection).

Note that you could try turning off the `automatic` mode for clog length setting and instead set a longer manual length...  If it was set longer to include the sync-feedback variability then it should still work, just not be as sensitive.

igiannakas, 2025-01-07:
Another thing to consider is to also use the tension sensor to detect clogs similarly to how BBL does it. If the rate of change (if using proportional belays, which I don’t think are a thing yet) or activation / de activation of the tension sensor is not within the control expected tolerance length wise (ie the sensor is in compression - extruder 50mm , the sensor should go into tension) trigger clog detection.