moggieuk/Happy-Hare #581: Encoder-corrected bowden move overshoots

Leonti, 2025-01-01:
Every so often the filement goes past the toolhead entry sensor and hits the extruder gears during the "correction" moves and I think it happens because of incorrect correction moves.

Here is my config from `MMU_STATUS SHOWCONFIG=1`:
```
Load Sequence:
- Filament loads into gate by homing a maximum of 70.0mm (gate_homing_max) to ENCODER
- Bowden is loaded with a fast CORRECTED 1500.0mm (calibrated_bowden_length) move, then homes to extruder using ENDSTOP 'extruder' and then moves 18.0mm (toolhead_entry_to_extruder) to extruder extrance
```

I use Filamentalist so I'm not surprised that the filament can sometimes slip when it's pulling fast directly from the spool, but I'm a bit confused about the correction moves.
Here is one example of a recent successful load (didn't hit the extruder gears, only relevant parts from the logs):
```
TRACE: Initial load into encoder. Stepper: 'gear' moved 70.0mm, encoder measured 57.0mm (delta 13.0mm). Pos: @70.0, (57.0mm)
TRACE: Fast loading move through bowden. Stepper: 'gear' moved 1443.0mm, encoder measured 1424.2mm (delta 18.8mm). Pos: @1513.0, (1493.7mm)
TRACE: Correction load move #1 into bowden. Stepper: 'gear' moved 18.8mm, encoder measured 18.4mm (delta 0.5mm). Pos: @1531.8, (1512.1mm)
DEBUG: Correction load move was necessary, encoder now measures 1512.1mm
TRACE: Correction load move #2 into bowden. Stepper: 'gear' moved 18.4mm, encoder measured 17.4mm (delta 1.0mm). Pos: @1550.2, (1529.5mm)
DEBUG: Correction load move was necessary, encoder now measures 1529.5mm
TRACE: (extruder sensor does not detect filament)
[T6] > En >>>>>>>>> (e) .. [Ex ... Nz] 1550.2mm (e:1529.5mm)
DEBUG: Homing to extruder 'extruder' endstop, up to 80.0mm
TRACE: Homing filament to extruder endstop. Stepper: 'gear' homed after moving 37.3mm (of max 80.0mm), encoder measured 37.7mm (delta -0.4mm). Pos: @1587.5, (1567.2mm). Pos: @1587.5, (1567.2mm)
DEBUG: Extruder endstop reached after 37.3mm (measured 37.7mm)
TRACE: (extruder sensor detects filament)
```

My understanding is that `mmu_calibration_bowden_length = 1500` starts from the encoder until a point close to the toolhead.
`Stepper: 'gear' moved 70.0mm, encoder measured 57.0mm` makes sense. First 13 mm of the movement were before the filament went into an encoder and 57mm were measured by the encoder.
We know that we already pushed 57mm of the filement past the encoder.
`Fast loading move through bowden. Stepper: 'gear' moved 1443.0mm` also makes sense. Since we already pushed 57mm we still need to push `1500 - 57 = 1443mm`.
`, encoder measured 1424.2mm (delta 18.8mm)` also makes sense, it tried to push `1443`, but encoder only measured `1424.2`, so we have `18.8mm` of slipped filament.
So now it does the correction move to bring it to the desired `1443` length:
`Correction load move #1 into bowden. Stepper: 'gear' moved 18.8mm, encoder measured 18.4mm (delta 0.5mm). Pos: @1531.8, (1512.1mm)`

Here is where the behavior is incorrect:
Since it detected that we are 18.8mm shy and it did the correction move of `18.4` as measured by the encoder it should be done already as it's supposed to be `1499.5mm` from the encoder entry.
Why is there a second correction move:
`TRACE: Correction load move #2 into bowden. Stepper: 'gear' moved 18.4mm, encoder measured 17.4mm (delta 1.0mm). Pos: @1550.2, (1529.5mm)`
I would have expected the correction move to be `0.5mm` mm if anything, but instead it seems like it's moving `18.4mm` which was the previous encoder measurement, and not `0.5mm` which was the delta.

I've looked at other corrections and the second correction move is always the encoder measurement of the first correction move which doesn't look right. I think it's the reason for overshooting, because it basically corrects double the amount of slippage.

Another thing which look strange is the positions:
```
moved 70.0mm, encoder measured 57.0mm (delta 13.0mm). Pos: @70.0, (57.0mm)
'gear' moved 1443.0mm, encoder measured 1424.2mm (delta 18.8mm). Pos: @1513.0, (1493.7mm) <- 70+1443=1513, 57+1424.2=1481.2 
'gear' moved 18.8mm, encoder measured 18.4mm (delta 0.5mm). Pos: @1531.8, (1512.1mm) <- 1513+18.8=1531.8, 1481.2+18.4=1499.6
'gear' moved 18.4mm, encoder measured 17.4mm (delta 1.0mm). Pos: @1550.2, (1529.5mm) <- 1531.8+18.8=1550.2, 1499.6+17.4=1517
```

The gear positions add up nicely and make sense, but the encoder positions are somehow different from the expected ones.

I'm still using `v2.7.2`, haven't had a chance to upgrade to `v3` yet.
Is it something that is addressed in `v3`?

Leonti, 2025-01-01:
Here are other examples:
```
00:42:17     TRACE: Correction load move #1 into bowden. Stepper: 'gear' moved 12.1mm, encoder measured 11.6mm (delta 0.5mm). Pos: @1525.1, (1512.1mm)
00:42:17     TRACE: Correction load move #1 into bowden. Stepper: 'gear' moved 12.1mm, encoder measured 11.6mm (delta 0.5mm). Pos: @1525.1, (1512.1mm)
00:42:17     TRACE: Correction load move #1 into bowden. Stepper: 'gear' moved 12.1mm, encoder measured 11.6mm (delta 0.5mm). Pos: @1525.1, (1512.1mm)
00:42:17   DEBUG: Correction load move was necessary, encoder now measures 1512.1mm
00:42:17   DEBUG: Correction load move was necessary, encoder now measures 1512.1mm
00:42:17   DEBUG: Correction load move was necessary, encoder now measures 1512.1mm
00:42:19     TRACE: Correction load move #2 into bowden. Stepper: 'gear' moved 11.6mm, encoder measured 11.6mm (delta 0.0mm). Pos: @1536.7, (1523.7mm)
00:42:19     TRACE: Correction load move #2 into bowden. Stepper: 'gear' moved 11.6mm, encoder measured 11.6mm (delta 0.0mm). Pos: @1536.7, (1523.7mm)
00:42:19     TRACE: Correction load move #2 into bowden. Stepper: 'gear' moved 11.6mm, encoder measured 11.6mm (delta 0.0mm). Pos: @1536.7, (1523.7mm)
```

```
00:15:43     TRACE: Correction load move #1 into bowden. Stepper: 'gear' moved 9.2mm, encoder measured 8.7mm (delta 0.5mm). Pos: @1522.2, (1512.1mm)
00:15:43     TRACE: Correction load move #1 into bowden. Stepper: 'gear' moved 9.2mm, encoder measured 8.7mm (delta 0.5mm). Pos: @1522.2, (1512.1mm)
00:15:43   DEBUG: Correction load move was necessary, encoder now measures 1512.1mm
00:15:43   DEBUG: Correction load move was necessary, encoder now measures 1512.1mm
00:15:45     TRACE: Correction load move #2 into bowden. Stepper: 'gear' moved 8.7mm, encoder measured 8.7mm (delta 0.0mm). Pos: @1530.9, (1520.8mm)
00:15:45     TRACE: Correction load move #2 into bowden. Stepper: 'gear' moved 8.7mm, encoder measured 8.7mm (delta 0.0mm). Pos: @1530.9, (1520.8mm)
```

Leonti, 2025-01-01:
Another interesting thing is that when I look at these logs:
```
00:03:49     TRACE: Initial load into encoder. Stepper: 'gear' moved 70.0mm, encoder measured 57.0mm (delta 13.0mm). Pos: @70.0, (57.0mm)
00:03:49     TRACE: (extruder sensor does not detect filament)
00:03:49 [T6] > En >........ (e) .. [Ex ... Nz] 70.0mm (e:69.6mm)
00:03:49   DEBUG: Loading bowden tube
00:04:26     TRACE: Fast loading move through bowden. Stepper: 'gear' moved 1443.0mm, encoder measured 1424.2mm (delta 18.8mm). Pos: @1513.0, (1493.7mm)
```

I don't understand how encoder position changed from `57mm` to `69.6mm`. `1443` is being calculated based on `1500` setting and `57mm` encoder position, but somehow when the movement finished the encoder position is not `57 + 1424.2 = 1481.2` as I would expect, but it's somehow `1424.2 + 69.6 = 1493.8` which corresponds to `Pos: @1513.0, (1493.7mm)`.
Not sure where discrepancy is coming from. Id's expect the position to be `Pos: @1513.0, (1481.2mm)`. `1513.0 = initial 70mm + additional 1443, 1481.2 = initial 57 + 1424.2 after the fast bowden move`

Leonti, 2025-01-02:
I believe I have fixed it with [this PR](https://github.com/moggieuk/Happy-Hare/pull/582).
I'm running it on my machine and after some testing I'm experiencing the overshooting issue anymore and usually a single correction move is now enough.

moggieuk, 2025-01-06:
Hi, I'll take a look at your PR this weekend, but to answer one question you posed:

The "calibrated bowden length" is the distance from the gate measurement point up to the extruder entrance.  The gate reference point used for calibration is also saved in `mmu_vars.cfg` so it can be adjusted if the gate homing point is changed (unlikely in most setups but I switch frequently in testing).    I know prior to v3.0 there was some ambiguity in exactly what the bowden length referred to.

moggieuk, 2025-01-06:
PR pulled.  Closing issue. Thanks.

closed 2025-01-06 by moggieuk