moggieuk/Happy-Hare #586: max_extrude_cross_section

PaulMellor, 2025-01-06:
Just Complete Latest Update "v3.0.1-63-g93d4fda8" , and am now getting "max_extrude_cross_section" errors which were not curing before. Check Gcode and Used a previously printed code but getting same error. Even tried a different Slicer? I cant explain why it all of the sudden is occurring.

"MMU issue detected. Print will be paused
Reason: Load sequence failed: Error running _MMU_POST_LOAD: Move exceeds maximum extrusion (20.905mm^2 vs 10.000mm^2)
See the 'max_extrude_cross_section' config option for details.
Occured when changing tool: > T0"

PaulMellor, 2025-01-09:
Sorry caught up elsewhere, Those values are still present in the mmu_macro_vars.cfg and are unchanged. I know they are also present in the Extruder config of the printer.cfg but, but has been in there for the last 3 backups going back into December.

So I am unsure why the mmu_macro_vars.cfg file is not overwriting the printer.cfg file.
Ok so the only thing I can think of that has changed is the order of the slicer scripts. as moved the scripts around slightly eg current sequence is mmu info >  the mmu start check > the printer start macro (which now includes the MMU_START_LOAD_INITIAL_TOOL) as I need the filament loaded to get a prime line. Going to move it back to the see if that corrects the issue.
Moved it into the start script at the very end as was finding on the load by the time it went through the print start it had oozed out leaving about a 5-10mm gap at the start of the print without filament.

ningpj, 2025-01-09:
From your klippy.log your [extruder] section is being loaded after mmu so last in takes precedence. I also saw you increased the value in your [extruder] section from 5 to 10 which was also reflected in later errors. It's only an upper limit so would simply update your extruder section to match or remove it altogether. It also relates to nozzle size in case you had this set higher before