moggieuk/Happy-Hare #587: Unknown config object 'mmu_machine' after 2.7.0 --> 3.0.1 upgrade

troypouliot, 2025-01-06:
I am getting this error after upgrading from 2.7.0 to 3.0.1. After the upgrade, I re ran the install.sh and made sure klipper is at the latest version.

This is the error from the klippy.log file,
`MMU: Hardware Initialization -------------------------------
Config error
Traceback (most recent call last):
  File "/home/pi/klipper/klippy/klippy.py", line 130, in _connect
    self._read_config()
  File "/home/pi/klipper/klippy/klippy.py", line 123, in _read_config
    self.load_object(config, section_config.get_name(), None)
  File "/home/pi/klipper/klippy/klippy.py", line 112, in load_object
    self.objects[section] = init_func(config.getsection(section))
  File "/home/pi/klipper/klippy/extras/mmu/__init__.py", line 21, in load_config
    return Mmu(config)
  File "/home/pi/klipper/klippy/extras/mmu/mmu.py", line 279, in __init__
    self._setup_mmu_hardware(config)
  File "/home/pi/klipper/klippy/extras/mmu/mmu.py", line 632, in _setup_mmu_hardware
    self.mmu_machine = self.printer.lookup_object("mmu_machine")
  File "/home/pi/klipper/klippy/klippy.py", line 79, in lookup_object
    raise self.config_error("Unknown config object '%s'" % (name,))
Error: Unknown config object 'mmu_machine'
webhooks client 1963820296: New connection
webhooks client 1963820296: Client info {'program': 'Moonraker', 'version': 'v0.9.3-4-ga4604e3'}`

My python is at version 2.7.16. I am hoping this isn't a Python 3 requirement thing...

ningpj, 2025-01-06:
HH configuration changed in 3.x and there's the mmu def moved from ``mmu_parameters.cfg`` to a ``[mmu_machine]`` configuration block in ``mmu_hardware.cfg`` which "may" not have been converted correctly by the installer.  
```
[mmu_machine]
num_gates: 12				# Number of selectable gates on MMU
mmu_vendor: Tradrack			# MMU family
mmu_version: 1.0e			# MMU hardware version number (add mod suffix documented above)
```
What does your ``mmu_hardware.cfg`` look like?

troypouliot, 2025-01-06:
Yeah, I'm not sure what happened. After going back and forth between 2.7.0 and 3.0.1 a couple times and uninstalling/reinstalling, I am back up and running on the current version. Thanks for the quick reply!

closed 2025-01-06 by troypouliot