moggieuk/Happy-Hare #588: MMU_CALIBRATE_BOWDEN BOWDEN_LENGTH depends on extruder_homing_endstop - working as intended?  wiki update needed?

jfurner, 2025-01-06:
In some testing recently I found the following behavior:

- If `extruder_homing_endstop: collision`, then during Bowden calibration `BOWDEN_LENGTH` needs to be shorter than the actual value (this is how the wiki describes it, measure the distance and subtract a bit from it).  In this case it performs a fast move at `BOWDEN_LENGTH`, then short moves until collision is detected.  Calibration fails if collision occurs during the fast `BOWDEN_LENGTH` move.

- If `extruder_homing_endstop: extruder`, then `BOWDEN_LENGTH` needs to be longer than the actual value.  In this case it performs a single fast move until the extruder switch is triggered.  Calibration fails if the switch is not triggered by the end of the fast `BOWDEN_LENGTH` move.  Following the wiki guide to subtract from the measured distance will result in calibration failure in this case.

If the above behavior is the intended function, could the wiki be updated to advise either adding or subtracting from the measured distance, depending on `extruder_homing_endstop` setting?

moggieuk, 2025-01-11:
@jfurner Thanks. The wiki is a little bit behind the codebase where enhancements have been made to the bowden calibration.  I have just even it a bit of an update:

https://github.com/moggieuk/Happy-Hare/wiki/MMU-Calibration#---step-5-calibrate-bowden-length

It would be great if you could review for clarity...

closed 2025-02-24 by github-actions[bot]

jfurner, 2025-08-12:
@AcrimoniousMirth can you share your `mmu_parameters.cfg` and a `mmu.log` from when the bowden length increasing is happening?  I'm particularly interested in `extruder_homing_endstop` and `toolhead_entry_to_extruder` parameters.  I wonder if your setup needs `toolhead_entry_to_extruder` to be set to 0 due to that delay.

AcrimoniousMirth, 2025-08-13:
I wouldn’t think it should be auto tuning either except it is, which leads me to believe there’s an un-exposed self correction that happens.

Unfortunately I’m also the author of the “Binky resolution 4mm” issue. Tried 2 different sensors, tried reprinting everything and it still consistently reads ~100 pulses for 400mm of movement. I’m wondering if the BTT MMB has a sample rate issue or something.
In either case, Binky’s been useless and I’ve disabled it now.