moggieuk/Happy-Hare #589: Instal happy hare without MMU

erwann17, 2025-01-07:
Good morning 
I would like to know if it is possible to use happy hare without MMU

I have a printer with 4 extruders and 1 hotend and I would like to configure it in multi filament 

all driven by a BTT Octopus

moggieuk, 2025-01-11:
You can install Happy Hare without MMU but it is designed to control filament loading / unloading of a single extruder.  It isn't designed to control multiple extruders.

Perhaps I'm confused about your question because it isn't clear what you want to do...

3DCoded, 2025-01-26:
Hi @erwann17 ,

If I'm understanding correctly, you're using four extruders, through a Y-splitter, to a single hotend?

If so, what you're doing is essentially a [3MS](https://github.com/3DCoded/3MS).

erwann17, 2025-01-27:
Hello, thank you for your feedback, this is exactly what I am looking for

closed 2025-03-12 by github-actions[bot]