moggieuk/Happy-Hare #592: Feature requests: Display encoder error/delta as a percentage.

Squid116, 2025-01-09:
When you move it gives you the desired move and the actual measured:

Take for example this result from `MMU_TEST_MOVE MOVE=100`:
```
1:09 pm Moved 600.0mm (measured 582.0mm)
```

It would be useful to keep an eye on that delta as a percentage i.e.
```
1:09 pm Moved 600.0mm (measured 582.0mm (97.0%))
```
or
```
1:09 pm Moved 600.0mm (measured 582.0mm (-3.0%))
```

ningpj, 2025-01-09:
Have you had a look at Happy Hare statistics as it tracks unload and load slippage per gate over time for this reason (e.g. data behind the gate status emojis)?  No problem if you already know about it, just wanted to make sure. Im also not near my printer so cant attach extra stats at the moment.

``mmu_stats detail=1``

https://github.com/moggieuk/Happy-Hare/wiki/Statistics-and-Consumption-Counters#---gate-statistics

moggieuk, 2025-01-11:
Indeed the "percentage" display in the stats is an aggregate % difference of movement vs measured movement.   I think this aggregate is more useful than individual % for h/w diagnostics.

The log messages for movement are quite distributed so adding a % to each move is a bit awkward.  Added to the "STEPPER" level debugging (`log_file_level: 4`) would be easy..

closed 2025-02-27 by github-actions[bot]