moggieuk/Happy-Hare #594: MMU_SYNC_GEAR_MOTOR SYNC=1 does not work out-of-print

norris108, 2025-01-10:
Despite calling `MMU_SYNC_GEAR_MOTOR SYNC=1` _before_ loading filament, after the filament has been loaded, the servo moves to the "up" position _before_ blobifier, which will inevitably fail

[mmu.log](https://github.com/user-attachments/files/18379584/mmu.log)

moggieuk, 2025-01-11:
Indeed, I see the problem:
When loading the bowden the extruder is correctly unsynced, but it isn't resynced until after the complete load operation.

This is probably a complete oversight in how I think about out-of-print sync state.  I'll have to think about how to address this.

moggieuk, 2025-01-11:
Patch pushed.  It seems to work as intended now. Please verify.

norris108, 2025-01-11:
Upgraded to `v3.0.1-68-g5d34de2c` and it seems to be working

closed 2025-01-11 by norris108