moggieuk/Happy-Hare #595: toolhead_unload_safety_margin doesn't appear to have any affect

norris108, 2025-01-10:
I've started having issue when unloading where the filament is stuck in the extruder

To mitigate this, i've increased `toolhead_unload_safety_margin`, however, it doesn't seem to unload any extra and is _still_ getting stuck

[mmu.log](https://github.com/user-attachments/files/18379715/mmu.log)

moggieuk, 2025-01-11:
I'm not seeing the cited problem in the supplied log...
```
16:15:47   DEBUG: Extracting filament from extruder
16:15:47     TRACE: _ensure_safe_extruder_temperature: current_temp=259.79, paused_extruder_temp=None, current_target_temp=260.0, klipper_minimum_temp=172.0, default_extruder_temp=260, source=auto
16:15:47   DEBUG: Unloading last 137.7mm to exit the extruder
16:15:56     TRACE: Unloading extruder. Stepper: 'extruder' moved -137.7mm, encoder measured 19.0mm (delta 118.7mm). Pos: @-183.0, (64.6mm)
16:15:56   DEBUG: Total measured movement: 19.0mm, total delta: 118.7mm
```

Because of no `toolhead` sensor or `extruder` sensor, HH uses "blind move" using just the extruder stepper (not synced) to pull the filament out of the extruder after the tip forming.  This log indicates it moved 137.7mm, which is a lot and I'm sure includes the `toolhead_unload_safety_margin`.   The fact that the encoder sees only 19mm of movement is telling...  
given the excessive extruder movement you would expect the encoder to read less but the big mismatch suggests that either the extruder is moving way more than necessary or the filament really is getting stuck in the extruder...

Anyway, HH is definitely moving 137.7mm.  Well, I say definitely because if the extruder isn't moving then it would indicate a very low level bug which I think is unlikely.   Best way to verify is to put your finger on the toolhead extruder stepper and see if you can feel this 137.7mm move....

norris108, 2025-01-12:
I increased `toolhead_unload_safety_margin` to some ridiculous value, `130`, to try and make sure the filament exits the extruder gears, but it *still* occasionally gets stuck

norris108, 2025-01-13:
Print quality had gone downhill today, so took my extruder apart to investigate and noticed my extruder tensioner arm had a crack in it so want applying enough pressure to the filament

Fixed that, and while I was there, added a toolhead sensor

Seems much more reliable now all-round, and my print quality has also improved significatly!

closed 2025-01-13 by norris108