moggieuk/Happy-Hare #596: EREC tried to cut filament whilst still loaded

norris108, 2025-01-10:
When filament is stuck in the extruder during unload, sometimes EREC tried to cut the filament whilst it's still loaded in the PTFE tube

[mmu.log](https://github.com/user-attachments/files/18379837/mmu.log)

moggieuk, 2025-01-11:
This is a tricky one..  the only sensor you have to validate filament in the bowden and beyond is the encoder...

(no gate sensor, no extruder entry sensor, no toolhead sensor, ...)

So HH does the following:
1. Does a synced short bowden move (i.e. it tries to ensure that the very end of the filament is not caught in extruder gears)
2. Defensively unloads the bowden.  Remember there is no gate sensor to home to so HH watches the encoder movement...  No (or tiny) movement means either: (i) the filament is stuck, (ii) the filament has exited the encoder at the gate.

It is impossible with just the encoder to determine if (i) or (ii) has occurred.  This is only of the reasons I advocate strongly for toolhead sensor (or if not a extruder entrance sensor).

JFYI: With a sync-feedback sensor it MAY be possible to detect a non moving filament IF the sync-feedback sensor design has a tension switch AND is sprung loaded to center position.  This isn't implemented in HH yet but on my radar.

norris108, 2025-01-13:
Added a toolhead sensor, so that should allow HH to detect when filament hasn't unloaded

closed 2025-01-13 by norris108

ningpj, 2025-01-13:
Welcome to the party :-)  Wondered when you would get around to swapping out your trusty CW1 or upgrading it with a toolhead sensor.   Many QOL improvement heading your way.

norris108, 2025-01-13:
Still Afterburner, but with [Poorman's SmartWatch](https://github.com/EtteGit/EnragedRabbitProject/tree/main/usermods/AB_Toolhead_Sensor/PoorMans_NEW_Smartwatch)!