moggieuk/Happy-Hare #597: Set default temperature range based on filament type

tkunchick, 2025-01-12:
***Feature request***
```
# ADVANCED: See documentation for use of these -----------------------------------------------------------------------
# ██████╗ ███████╗███████╗███████╗████████╗    ██████╗ ███████╗███████╗███████╗
# ██╔══██╗██╔════╝██╔════╝██╔════╝╚══██╔══╝    ██╔══██╗██╔════╝██╔════╝██╔════╝
# ██████╔╝█████╗  ███████╗█████╗     ██║       ██║  ██║█████╗  █████╗  ███████╗
# ██╔══██╗██╔══╝  ╚════██║██╔══╝     ██║       ██║  ██║██╔══╝  ██╔══╝  ╚════██║
# ██║  ██║███████╗███████║███████╗   ██║       ██████╔╝███████╗██║     ███████║
# ╚═╝  ╚═╝╚══════╝╚══════╝╚══════╝   ╚═╝       ╚═════╝ ╚══════╝╚═╝     ╚══════╝
#
# Examples...
# Gate:                #0      #1      #2      #3      #4      #5      #6      #7      #8
#gate_status:          1,      0,      1,      2,      2,     -1,     -1,      0,      1
#gate_filament_name:   one,    two,    three,  four,   five,   six,    seven,  eight,  nine
#gate_material:        PLA,    ABS,    ABS,    ABS+,   PLA,    PLA,    PETG,   TPU,    ABS
#gate_color:           red,    black,  yellow, green,  blue,   indigo, ffffff, grey,   black
#gate_temperature:     210,    240,    235,    245,    210,    200,    215,    240,    240
#gate_spool_id:        3,      2,      1,      4,      5,      6,      7,      -1,     9
#gate_speed_override:  100,    100,    100,    100,    100,    100,    100,    50,     100
#endless_spool_groups: 0,      1,      2,      1,      0,      0,      3,      4,      1
#
# Tool:                T0      T1      T2      T3      T4      T5      T6      T7      T8
#tool_to_gate_map:     0,      1,      2,      3,      4,      5,      6,      7,      8

# SUPPLEMENTAL USER CONFIG retained after upgrade --------------------------------------------------------------------
#
```

there is not much information [here](https://github.com/moggieuk/Happy-Hare/wiki/Configuring-mmu_parameters.cfg#---statically-defined-reset-defaults)

What I am attempting to accomplish is setting temperature ranges for different filament types for loading and unloading purposes.

such as 
```
#gate_material:           PLA,    ABS,    ASA,    PETG,   TPU
#gate_load_temperature:   210,    240,    265,    235,    210
#form_tip_temperature:    210,    190,    225,    215,    190
```

basically, if these filaments are installed, it would use these temperatures instead of a hard coded temp.

moggieuk, 2025-01-14:
You can do that with the `MMU_GATE_MAP` command.  The above parameters are simply those used on "reset" and if not defined the reset value are hard coded - temp will be the defined `default_extruder_temp`

E.g.
```
MMU_GATE_MAP GATE=0 TEMP=xxx
```
This will manual set the default temp for the filament in gate 0.  Note that generally temperatures are set by the slicer. These are read into the slicer_tool_map (available via printer variables) and can be used in custom macros if you desire

Read this wiki page: https://github.com/moggieuk/Happy-Hare/wiki/Tool-and-Gate-Maps

tkunchick, 2025-01-14:
I don't think I was clear on what I am trying to accomplish.

I did see the spoolman topic, though doesn't quite meet my goals. So far, I am hand writing on the spools what ID each is.

I said loading, but I am more trying to change the tip forming temperatures for different filament types.

I did see that the loading temperatures are pulled well from the slicer, so I guess I am just looking to change the tip forming temps, as some do better at different temps from what I have seen so far.

Tip forming is a bit of a PITA haha.

closed 2025-02-27 by github-actions[bot]