moggieuk/Happy-Hare #598: moonraker complains about "mmu server" config section

BennyBenasy, 2025-01-15:
because of this error there is a problem with logging into the web interface. logging in takes 5-10 attempts. how to fix this?
[image]

ningpj, 2025-01-15:
Please ssh to your pi and rerun the Happy Hare installer without arguments to make sure all the python modules are in the right place, reboot your pi and try again. This is basically indicates moonraker cant find the Happy Hare mmu_server python module so doesn't understand the ``[mmu_server]`` section in its config. 

```
cd ~/Happy-Hare
./install.sh
```
The Happy Hare discord is the best place for quick advice, workarounds, etc and leveraging the collective experience of the broader community. Github is best suited for reporting bugs or enhancements.  Also when reporting issues or reaching out for help, please include as much detail you can such as the versions of HH, Klipper, Moonraker you are using, steps you have taken to diagnose or work around the issue, etc.  Logs like ``mmu.log``, ``klippy.log``, and ``moonraker.log`` are especially relevant.   

You can join the dedicated Happy Hare forum if you haven't already: https://discord.gg/98TYYUf6f2

BennyBenasy, 2025-01-15:
> Пожалуйста, подключитесь по ssh к вашему pi и перезапустите установщик Happy Hare

this will not reset my customized mmu configs?

BennyBenasy, 2025-01-15:
Yes, this fixed the error and my configs were not lost. Many thanks

closed 2025-01-15 by BennyBenasy

ningpj, 2025-01-15:
Excellent.  You can run it without arguments without loss of configuration unless its a major update the requires manual intervention.  Its also recommended to run it manually rather than just relying on the moonraker update manager as its not always able to apply the required changes.  It will also download and apply updates.