moggieuk/Happy-Hare #599: MMU disabled - Last gate stepper still synced

tkunchick, 2025-01-16:
Type B MMU

I am trying to totally bypass the MMU and just print from a drybox, I ran `MMU ENABLE=0`, but the last active gate is still trying to push filament.

Is there a better bypass method for type B MMUs?

moggieuk, 2025-01-17:
Ah. Good catch.  This should unsync() on disable.  I will fix.
There is another way, but I have tested on type-B MMU's yet:

In `mmu_hardware.cfg`
```yml
[mmu_machine]

:

has_bypass: 1
```

This should enable the bypass functionality built into HH. You can then "select" bypass with:
```
MMU_SELECT BYPASS=1
```
On type-A MMUs this moves selector to a designated position and turns of most of the MMU functionality.  It should so the same with type-B but I haven't tested.  I could imagine having and extra free bowden going into the splitter/combiner which represents the "MMU bypass".

Let me know how this works out.

tkunchick, 2025-01-17:
that worked. 

just had to recover the gate that was previously loaded to unloaded before activating bypass

```
MMU_RECOVER LOADED=0
MMU_SELECT BYPASS=1
MMU ENABLE=0
```

tkunchick, 2025-01-17:
off topic, but I been messing around with the Mainsail popup prompts, didn't know if you wanted to have a look.

probably a better way to do it based on the mmu setup already, but I just tinker with coding.. not a programmer :-)

[_my_mmu_gcode.cfg.zip](https://github.com/user-attachments/files/18457594/_my_mmu_gcode.cfg.zip)

moggieuk, 2025-01-19:
I will have a look. Note that @ammmze  was also working on something similar (at least I think so without yet looking at what you have done!).  I think it is a GREAT idea (I just don't have time to help right now).

I'm adding the unsync() to disable now.  Will be in the `multi_mmu` branch which will be pushed to `main` in the next 1-2 weeks.

ammmze, 2025-01-19:
> Note that @ammmze was also working on something similar

Yep...still needs built out, but was implementing it in a way that it would be done from python and abstract away the gcode aspects of the prompt using some async coding. It makes for a unique way of doing the prompts but I think it makes it easier to manage the many possible prompts and errors and what not.

It's been a couple months since I've touched it, but [here's](https://github.com/ammmze/Happy-Hare/commit/8420e0331aadda25825d33f0f73b55350dc0db09) what I have so far.

tkunchick, 2025-01-19:
@ammmze 

yeah, way above me. I am decent in batch, vbscripting and Powershell (MS windows admin for 30+ years)... getting better with gcode macros... but python has alluded me so far.

I can read it and get the jest of it.. but that's about it.

I know the way I did it would be a lot of work giving the various HH configurations and MMU configurations. reading directly from HH would clearly be a better solution.

just thought it was neat and thought I would share.

tkunchick, 2025-01-20:
> that worked.
> 
> just had to recover the gate that was previously loaded to unloaded before activating bypass
> 
> ```
> MMU_RECOVER LOADED=0
> MMU_SELECT BYPASS=1
> MMU ENABLE=0
> ```
@moggieuk 
Now I get an odd message when the printer needs to pause for anything.
```
10:36 AM
Error running __PAUSE: {"code":"key585","msg":"Move out of range: 306.843 156.076 34.183 [2368.812]", "values":[306.843, 156.076, 34.183, 2368.812]}
10:36 AM
{"code":"key585","msg":"Move out of range: 306.843 156.076 34.183 [2368.812]", "values":[306.843, 156.076, 34.183, 2368.812]}
10:36 AM
{"code":"key585","msg":"Move out of range: 306.843 156.076 34.183 [2368.812]", "values":[306.843, 156.076, 34.183, 2368.812]}
10:36 AM
{"code":"key585","msg":"Move out of range: 306.843 156.076 34.183 [2368.812]", "values":[306.843, 156.076, 34.183, 2368.812]}
10:36 AM
Parking toolhead at (x:306.0, y:156.0, z:34.0) for pause operation
```
These locations are perfectly find when running with the MMU on. I did notice that the x is different than the config of 306. I believe the max on the K1 Max is 306.5mm
```
variable_park_toolchange        : 306, 156, 1, 5, 2	; x,y,z-hop,z_hop_ramp,retract for "toolchange" operations (toolchange,load,unload)
variable_park_runout            : 306, 156, 1, 5, 2	; x,y,z-hop,z_hop_ramp,retract
variable_park_pause             : 306, 156, 5, 0, 2	; x,y,z-hop,z_hop_ramp,retract (park position when mmu error occurs)
variable_park_cancel            : 306, 156, 10, 0, 5	; x,y,z-hop,z_hop_ramp,retract
variable_park_complete          : 306, 156, 10, 0, 5	; x,y,z-hop,z_hop_ramp,retract
```
[klippy.log](https://github.com/user-attachments/files/18480070/klippy.log)

[mmu.log](https://github.com/user-attachments/files/18480074/mmu.log)

closed 2025-03-05 by github-actions[bot]