moggieuk/Happy-Hare #602: Enhancement: Allow `has_filament_buffer` to be configured per gate

burkfers, 2025-01-19:
The upcoming `has_filament_buffer` addition is a great option to make setting up for eg Filamentalist easier.

If one could configure this per gate, it would allow true flexibility for hybrid builds, so that the buffered gates can load and unload at high speed, yet the heavy spool on the rewinder-equipped gates can be moved slowly with the required torque.

moggieuk, 2025-01-26:
Filing as feature request -- you never know, but after Mailsail support!

theodiem, 2025-02-01:
+1 for it (subscribing to the feature request). 

My usercase is that I have a drybox which doesn't fit filamentalists inside, so gates 0-3 have buffers (also to use with TPU) and the remaining gates I use filamentalist for passive rewinding mechanism.

moggieuk, 2026-10-23:
Thanks @burkfers, the per-gate idea is acknowledged. Related work is now underway in #856: per-gate spool weight in the gate map, driving espooler power. That doesn't add a per-gate `has_filament_buffer`, so I'm keeping this open, but it's a step towards per-gate tuning for heavy spools on rewinder gates.

moggieuk, 2026-10-23:
Update: the related work in #856 has been merged into `development` (#1450) and will be released in **v4.2.0**. The gate map now carries a per-gate spool weight, and espooler power can scale with how full each spool is. This doesn't add a per-gate `has_filament_buffer`, so I'm leaving this issue open for that.