moggieuk/Happy-Hare #603: Sync unload move option

Blendan1, 2025-01-19:
When unoading fron the toolhead i have the issue that (even with as best as i can get it) tip forming sometimes the extruder has a very slight grip on the fillament. That then stops the unloading move.

What i would loike to see is that when unloading from the extruder and you have sync enabled it also uses the gear for the toolhead unload.

Blendan1, 2025-01-20:
I had a look in the code and I think it's a bug:

Sync and sync for tip form are both enabled.

Expected:

Print end -> Unload called -> tip from (synced) -> extruder unload (synced) --> unload etc (not synced)

What happened:

Print end -> Unload called -> tip from (synced) -> extruder unload (not synced) --> unload etc (not synced)

I would expect if sync is enabled and extruder unload is called, without extruder only, then it will always be synced. I changed the mmu.py locally so that the sync variable is set true when  it is not extruder only without any additional checks, that results in the expected behaviour for me.

My guess is that something in the sync check is not working as it should, sadly I don't know enough about the project to be able to fix it myself.

vajonam, 2025-01-23:
I too am observing this. I will collect some better logs with debug it show when/what is going on exactly glad I am not alone in t his.

thisiscam, 2025-02-07:
Didn't find this thread before but here is my observation and log, that seem related as well: https://github.com/moggieuk/Happy-Hare/issues/624

Blendan1, 2025-03-04:
I think the issue here is that it checks the syn state, but from what i understand it should only check if motor sync is enabled and its not `extruder_only`
https://github.com/moggieuk/Happy-Hare/blob/57f94e6bd23bb52a6753bd9b0fbdb6a4030ba299/extras/mmu/mmu.py#L4486

or there should be an extra option like with the tip forming where you can set if unloading should be done synced (if not `extruder_only`)

I changed my own mmu.py localy and it unloads synced, but in the newest version it then looses track of the filament position and slowly unloads.
(though that might also be my fault but i haven't been able to track down why it happens)

vajonam, 2025-03-04:
I haven't seen this with the latest 3.1 versions. just FTI.

Blendan1, 2025-03-04:
@vajonam also while not printing or only while printing? my issue is also always in my PRINT_END macro.

vajonam, 2025-03-04:
To be honest, everything is just working I (famous last words), so I haven't been paying attention to the system. I will have a look on the next couple of prints during unload.

Blendan1, 2025-03-31:
I forgot to comment, but in version v3.1.0-101 i still have the issue.

closed 2026-10-21 by moggieuk