moggieuk/Happy-Hare #604: feature request: add message and reason the

vajonam, 2025-01-20:
is it possible to send the message and reason similar to `_MMU_ERROR_DIALOG`, when using `variable_user_mmu_error_extension` ? Looking through the macros I dont see _MMU_ERROR ever being called which then calls that macro. I am happy to work on  a PR if provided some direction.

vajonam, 2025-01-20:
is this where I add 

https://github.com/moggieuk/Happy-Hare/blob/6853243cceb19c6711508fe0bb840f6b9b39f187/extras/mmu.py#L2559

Something like 

`self._wrap_gcode_command("_MMU_ERROR MSG='%s'" REASON='%s' % (msg,reason))`

and then modify the _MMU_ERROR gcode to send the appropriate 

`{vars.variable_user_mmu_error_extension} MSG={MSG} REASON={REASON}` 

will this work? if this is correct. I will provide a PR, please let me know if there anything I have missed.

moggieuk, 2025-01-26:
This is used during the startup macros and is automatically called on `MMU_PAUSE` (i.e. a MMU error occurs, directly calling "MMU_PAUSE" is a fake error).

Internally this will call the `error_dialog_macro` if `show_error_dialog` is true.  

```
MMU_PAUSE MSG="my custom message"
```
might be all you need...

vajonam, 2025-01-26:
My end goal is to be able to send a mqtt notication using a macro that I have already defined. Like this.

`_SEND_MMU_ERROR_NOTIFICATION PAYLOAD={param.MSG}` 

I am not understanding how the above to help me pass the message to an an arbitrary macro.

closed 2025-03-11 by github-actions[bot]