moggieuk/Happy-Hare #608: FR: Number gate sensors with two digits so they sort correctly in the Mainsail interface

NameOfTheDragon, 2025-01-24:
In Mainsail, my gate sensors appear thus:

[image]

It is a friction point that gates 10 and 11 are out of order.

Possible solution? Make all the numbers have 2 digits, i.e. add leading zeroes to the names of sensors 0 through 9, so that they sort correctly when using an alphabetical sort.

I'm just guessing that alphabetical sort is the problem; there may well be a better way of resolving this.

moggieuk, 2025-01-26:
Yes, that would be a solution.  I'll mark as a feature request..  I'm usually pretty good at picking up requests when I'm in that area of the code...

moggieuk, 2026-10-21:
This is now resolved in Mainsail itself, so no Happy Hare change is needed.

Mainsail v2.17.0 (January 2026) switched its sensor list to a natural sort (mainsail-crew/mainsail#2380), so `mmu_entry_9` now lists before `mmu_entry_10`. If you still see the old order, updating Mainsail will fix it.

For reference, Fluidd doesn't sort filament sensors; it lists them in the order Klipper registers them, which is already gate order. KlipperScreen (Happy Hare Edition) hides the per-gate sensors entirely.

We looked at zero-padding the sensor names (`mmu_entry_09`) instead, but with the UIs already handling this it would only rename sensors on larger machines (losing any hidden-sensor preferences saved by name) without changing what anyone sees on a current Mainsail.

closed 2026-10-21 by moggieuk