moggieuk/Happy-Hare #610: MMU issue: Unload sequence failed: Error running _MMU_CUT_TIP

Culem94, 2025-01-25:
I have Voron 2.4 350mm, ERCF V2 and filamentrix cutter. My issue appears, when printer trys to cut filament with mmu_cut_tip macro.

17:52:17 Forming tip...
17:52:17 Automatically homing XY
17:52:22 Warning: Klipper reported out of range gcode position (x:350.03251015625, y:352.00309609375)! Adjusted to (x:350.0, y:352.0) to prevent move failure
17:52:22 Retracting filament 29.0mm prior to cut
17:52:22   DEBUG: Restoring pressure advance: 0.030000
17:52:27 Unsynced MMU from extruder
17:52:27 MMU issue: Unload sequence failed: Error running _MMU_CUT_TIP: Move out of range: -305.000 352.000 5.000 [-29.000].
         Occured when unloading tool

I found out that if I put zero in this settings (mmu_macro_vars.cfg):

-variable_pin_loc_xy        : 0, 352	; x,y coordinates of depressor pin
-variable_pin_park_dist    : 0		; Distance in mm

there is no error from above, but printer head obviously cut filament to ˝early˝. If I replace form_tip_macro: _MMU_CUT_TIP from mmu_parameters.cfg file, with _MMU_FORM_TIP, there is also no error. 
How I run in this error? In the evening, I was printing model with 3 colors. When it was over I shutdown printer as usually. Next day I turn on printer and start printing new model with 2 colors.  When the printer wanted to change color, there was this error. I didn't change any settings. Problem is still there also if I try to unload manual.

I do not know, if this is bug or problem with my configuration.

P.S. sorry for bad english

moggieuk, 2025-01-26:
Try homing first rather than relying on the automatic home -- I bet it will work.

 wonder if, when the macro "corrects what klipper reports" it is getting confused.  I'll add to the list of bugs to look at.

Culem94, 2025-01-26:
I try home first and then unload, but is it the same error

moggieuk, 2025-02-05:
Can you attach your printer configuration and/or `klippy.log`.  I can see if I can figure out what might be going on..

Culem94, 2025-02-05:
[mmu.log.txt](https://github.com/user-attachments/files/18667077/mmu.log.txt)

[config-20250205-062310.zip](https://github.com/user-attachments/files/18667079/config-20250205-062310.zip)

[klippy.log (1).txt](https://github.com/user-attachments/files/18667082/klippy.log.1.txt)

Hi, this is my configs and logs
I will make this debug later

Culem94, 2025-02-05:
> I can't reproduce, so you will have to debug a little for me... Carefully edit `mmu_cut_tip.cfg` and replace this macro with this debug version. This simply adds a few `M118` commands to log what it is doing (you can remove the M118 lines later to restore to original):
> 
> ###########################################################################
> # Helper macro for tip cutting
> #
> [gcode_macro _FILAMETRIX_MOVE_TO_CUTTER_PIN]
> description: Helper to move the toolhead to the target pin in either safe or faster way, depending on toolhead clearance
> gcode:
>     {% set pin_park_x_loc = params.PIN_PARK_X_LOC|float %}
>     {% set pin_park_y_loc = params.PIN_PARK_Y_LOC|float %}
>     {% set vars = printer['gcode_macro _MMU_CUT_TIP_VARS'] %}
> 
>     {% set safe_margin_x, safe_margin_y = vars.safe_margin_xy|map('float') %}
>     {% set travel_speed = vars['travel_speed']|float %}
> 
>     M118 _FILAMETRIX_MOVE_TO_CUTTER_PIN: X{pin_park_x_loc} Y{pin_park_y_loc} safe_margin_x={safe_margin_x}, safe_margin_x={safe_margin_y}
>     {% if ((printer.gcode_move.gcode_position.x - pin_park_x_loc)|abs < safe_margin_x) or ((printer.gcode_move.gcode_position.y - pin_park_y_loc)|abs < safe_margin_y) %}
>         # Make a safe but slower travel move
>         M118 G1 X{pin_park_x_loc} F{travel_speed * 60}
>         M118 G1 Y{pin_park_y_loc} F{travel_speed * 60}
>         G1 X{pin_park_x_loc} F{travel_speed * 60}
>         G1 Y{pin_park_y_loc} F{travel_speed * 60}
>     {% else %}
>         M118 G1 X{pin_park_x_loc} Y{pin_park_y_loc} F{travel_speed * 60}
>         G1 X{pin_park_x_loc} Y{pin_park_y_loc} F{travel_speed * 60}
>     {% endif %}
> Re-run and share the logs with me.

This mmu cut tip.cfg I need to find in RPI maps and there I can change this debug version, yes?

vajonam, 2025-02-05:
I have this in now, will let you know how this goes. I have my happy hare and ercf tuned, this used to happen a bunch when I was testing testing the cut using the test macro from a cold start. i.e. make some changes, restart klipper and then run `MMU_TEST_FORM_TIP`  macro. I have debug in there there but like said I am happy with my tuning  of parameters. also I haven't seen it a while. but I have made too many changes, moved to multi-mmu branch, fixed up my tuning etc. so hard to tell what changed.

Culem94, 2025-02-05:
> I can't reproduce, so you will have to debug a little for me... Carefully edit `mmu_cut_tip.cfg` and replace this macro with this debug version. This simply adds a few `M118` commands to log what it [...]

Hi. I made this operation with debug configuration.

[klippy (3).log](https://github.com/user-attachments/files/18675082/klippy.3.log)
[mmu (5).log](https://github.com/user-attachments/files/18675081/mmu.5.log)

moggieuk, 2025-02-15:
I'm sorry I wasn't clear. The M118 will write to the console log.  Can you send that when you experience the error...

Culem94, 2025-02-15:
> I'm sorry I wasn't clear. The M118 will write to the console log. Can you send that when you experience the error...
> 
> I've looked again at the logs you supplied and everything looks fine, so I need the console output with debugging to go the next step..

[image]

what am I doing wrong?

Culem94, 2025-02-15:
> I'm sorry I wasn't clear. The M118 will write to the console log. Can you send that when you experience the error...
> 
> I've looked again at the logs you supplied and everything looks fine, so I need the console output with debugging to go the next step..

[image]

Culem94, 2025-03-17:
> I'm sorry I wasn't clear. The M118 will write to the console log. Can you send that when you experience the error...
> 
> I've looked again at the logs you supplied and everything looks fine, so I need the console output with debugging to go the next step..

Is this good now?

[image]

closed 2025-05-15 by github-actions[bot]