moggieuk/Happy-Hare #611: Question about the client_macros PAUSE

Sparkss, 2025-01-26:
I had a pause fail to set the long timeout and blow out a long print near the end, but it was called by blobbifier and I already worked around that. That caused me to search through the various pause macros and I noticed in the mmu client macros pause that the _MMU_RESTORE_POSITION only gets called if mmu is not enabled in the printer config. Is this correct?

[gcode_macro RESUME]
rename_existing: BASE_RESUME
description: Resume the print
gcode:
    {% set vars = printer['gcode_macro _MMU_CLIENT_VARS'] %}

    {% if not printer.pause_resume.is_paused %}
        MMU_LOG MSG="Print is not paused. Resume ignored"
    {% else %}
        {vars.user_resume_extension|default("")}
        {% if not printer.mmu.enabled %}
            _MMU_RESTORE_POSITION	# This will take the correct "over and down" movement path and unretract
        {% endif %}
        BASE_RESUME
    {% endif %}

Thanks in advance.

moggieuk, 2025-02-04:
`_MMU_RESTORE_POSITION` is correctly called -- it restore the toolhead position and since it is a HH feature is disabled when the MMU is disabled (essentially turning HH off).

As for timeout... HH is controlled with:
```yml
timeout_pause: 72000            # Idle time out (printer shuts down) in seconds used when in MMU pause state
disable_heater: 600             # Delay in seconds after which the hotend heater is disabled in the MMU_PAUSE state
```
This set `idle_timeout` when the MMU goes into an error state, but NOT if `PAUSE` is called directly.   Does blobifier call PAUSE? if so it should call MMU_PAUSE instead.. Will check and talk to @Dendrowen

Dendrowen, 2025-02-04:
Yes, an oversight. Excuse me!

moggieuk, 2025-02-04:
Ok, adding:
```
    M117 Empty purge bucket!
    MMU_PAUSE MSG="Empty purge bucket!"
```

moggieuk, 2025-02-04:
Fixed.

closed 2025-03-21 by github-actions[bot]