moggieuk/Happy-Hare #612: Orcaslicer + klipper +Spoolman+ happy-hare

Romano26, 2025-01-30:
Hi
i need help, i use orcaslicer with spoolman for filament management.
i used the tutorial to assign the filament to the gate but it doesn't work, no change

[image]

does it work for you ?

moggieuk, 2025-02-05:
If you run `MMU_GATE_MAP` do you see the spoolID on each gate? E.g.
```yml
mmu_gate_map
Gates / Filaments:
Gate 0: Status: Spool, SpoolId: 1 --> Material: PLA - Silk, Color: 8cdfac, Name: Matte Green, Temp: 210
Gate 1: Status: Spool, SpoolId: 2 --> Material: PTEG, Color: dcda34, Name: eMarble, Temp: 220
```

If so then filament details are being pulled from spoolman.  If not and you are sure you set the `spoolman_support` parameter? It has to not be "off" for this to work.

Finally, if you want spoolman to be the source of truth (unusually unless in print farm) then you would set `spoolman_support: pull` and set the gate assignment in spoolman database, either directly or with the `MMU_SPOOLMAN` command (which is a direct write to the DB).

igiannakas, 2025-02-07:
I’ve been using orca and spool man for a while now.

For me the simplest way to set it up was to have spoolman be the source of truth, setup HH in pull mode and manually set the filaments in the orca UI match the printer and gate allocation in spoolman. Thatway using the Ui I can reassign filament to gates etc.

There may be a better way to do this but I’ve found this to be the most intuitive for how my brain works :)

Romano26, 2025-03-17:
Thanks for your help.
The spoolman parameters are correct; they're on pull.
When I run MMU_GATE_MAP, everything is empty, with no information.
I want to use Orca to assign the correct filament with the spoolman ID.
Orca correctly assigns the correct ID to spoolman in Mansail, but it doesn't work in Klipperscreen.
Do I need to add information to the filament startup G.code?

closed 2025-04-30 by github-actions[bot]