moggieuk/Happy-Hare #616: Wiki: Correct macro name on Slicer setup

trajesus, 2025-02-03:
On the wiki, the slicer setup states that the last macro on the end Gcode should be _MMU_END_PRINT, however, the macro is called _MMU_PRINT_END and, as such that should change in the wiki to avoid confusion.

When following the guide, as is, the print finishes with an error of unknown macro "_MMU_END_PRINT".

moggieuk, 2025-02-04:
Thanks, fixed.   Actually the direct macro name is now "MMU_PRINT_END" (the `_` is for backward compatibility).  Note that you can completely omit this if you are not using Octoprint.  Harmless to retain but not necessary since HH sees the end of print hook and calls internally anyway.

trajesus, 2025-02-05:
Makes sense, thank you. I'll close the issue as it is solved.

closed 2025-02-05 by trajesus