moggieuk/Happy-Hare #618: Selector touch blocked filament recovery

Thomsen1982, 2025-02-03:
Unfortunately, this feature no longer works. Although it recognizes that a filament is blocked, it does not try to deposit the filament again. Instead, it is loaded again and you have the wrong color.
That used to work very well, but unfortunately not anymore.
It would be nice if there were fewer new functions, but they would be properly tested.
The happy hare would otherwise work so well, praise for that, but unfortunately bugs regularly creep in with functioning functions.

[mmu.log](https://github.com/user-attachments/files/18648063/mmu.log)

moggieuk, 2025-02-04:
I will have a look but but logic hasn't changed in many releases.   What I see in the log is that you have the "automatic retry" enabled -- I personally don't recommend this because it often hides the real issue. I.e. it carries on blindly after error.

Also, assuming that your encoder is working well, I see this pattern:
```
11:12:15   DEBUG: Selecting tool T0 on gate 0...
11:12:17 Selector is blocked by filament inside gate, will try to recover...
11:12:17   DEBUG: Checking for filament at gate...
11:12:17 Attempting to recover filament position...
11:12:17   DEBUG: Checking for filament in MMU...
11:12:18 > MMU_STATS COUNTER=servo_down INCR=1
11:12:19   DEBUG: Filament not detected in encoder after buzzing gear motor
11:12:19   DEBUG: Filament already ejected
```
The "blockage" and then "filament not detected" suggests that stallguard is not tuned correctly -- it appears to be firing when the gate isn't blocked.  This could be because of new friction in the movement of the selector cart...(?)

To diagnose further I'd recommend turning off the automatic retry, also try running a selector soaktest: `MMU_SOAKTEST_SELECTOR HOME=1`

Thomsen1982, 2025-02-05:
Thank you for the hints.
I deactivate the automatic retry and run the test 
I think at the weekend I can check it.

The stallguard was tuned realy good, I checked it.
But you are right, it could be because of a new friction.

Thomsen1982, 2025-02-10:
I have now made a longer print. No idea what was going on, errors without end.
everything was set up so that I was able to make prints with over 1000 changes. i'm a long way off that now.

the constant errors that there was no filament, strange.
i also had the autotune functions on to see what they were doing, maybe it was also something that was messing me up.

you could check one type of error to see if there is something there. i mean the errors with the external filament. there were a few, but only one was recognized correctly. all the others were the filament in the selector. they should have been corrected, but were recognized incorrectly.

the whole behavior seems to be quite strange at the moment, so i'm going to remove happy hare completely and set it up fresh. hopefully i'll get back to reliability then

greetings
thomsen

[mmu.log](https://github.com/user-attachments/files/18729052/mmu.log)

salocain, 2025-02-18:
I have exactly the same problem , but i see it's more in thé middle of m'y 12 gates ercf ,i think the rod guide of selector , arc ,and less grip , maybe an option to configure a différents servo angle it's a good idea.

I don't understand the Logic of unload/reload on selector error.

Filament unload --> codeur didn't see mouvement (but a little part of filament stay --> selector move and blocage are detexted immediately , ercf unload but hé unload many mm and the filament go behind the gear...
Thé selector homing -> go to next gate ,etc ,but when ercf come back to the gate where filament is behind the gear , i need to push the filament...

That is the problem.

Thomsen1982, 2025-02-28:
So, unfortunately it took a while. But now I've got around to it.

I have completely reinstalled Happy Hare, as well as the config files. Afterwards I only transferred my values to the new config files by comparison.
I noticed that some names have changed. That is certainly not so clever. The update does not seem to be reliable.

I started a few short prints. So far everything has worked, as has the recovery after a problem. I noticed that without changing anything on the hardware and with the same values in the config.

So this is more of a software problem, which could perhaps be caused by the many updates, as this is not carried out properly. Perhaps an error is also building up, as quite a lot is saved and this is also partially adjusted automatically.

The bottom line is that everything seems to be working so far. 
I have now deactivated updates and will transfer the config to the new config files by comparison if necessary.

This closes the problem.

Thomsen1982, 2025-02-28:
> I have exactly the same problem , but i see it's more in thé middle of m'y 12 gates ercf ,i think the rod guide of selector , arc ,and less grip , maybe an option to configure a différents servo [...]

In my case, the errors were gone again after a short time.

If the sector works inaccurately or irregularly for you, perhaps a little tip.
In the selector I have a bearing with a V cut-out. I have shrink tubing over it to increase the tension. Maybe that will help.

Otherwise, set it up fresh. The setup wizard is really good

salocain, 2025-02-28:
After a modification of the selector  hold current , problème desapear , I think the filament in the load/unload sequence move a little bit the selector , and after many sequence , the selector isn't at the right position.

I put 0,4a on the hold current