moggieuk/Happy-Hare #620: Allow for duplicate entry/exit led definition

ripnet, 2025-02-04:
Low priority request, but I have duplicated LEDs in my MMU and I would love to be able to light them all up. There are "exit" LEDs on both the spool rewinders and on the skirt of the machine itself.

I am asking for something like:

```
[mmu_leds]
exit_leds:    neopixel:mmu_leds_1 (2-5)
              neopixel:mmu_leds_2 (1-5)
entry_leds:   neopixel:mmu_leds_3 (6-10)
              neopixel:mmu_leds_3 (1-5)
exit_leds:    neopixel:mmu_leds_3 (16-20)
              neopixel:mmu_leds_3 (11-15)
frame_rate: 24
```

Right now, it seems that the ones that were last defined take precedence and the early ones don't turn on.

moggieuk, 2025-02-04:
I do see an error in this setup BTW (maybe just a type)
I see 9 leds on initial `exit`
but 10 on on `entry` and second `exit`

I'll add as feature request but it will be low priority... you obviously love LEDs. Lol.  Technically the LED support is all done in macros (after h/w setup which can also be done) so you could add this yourself...

ripnet, 2025-02-05:
Good catch -- I fixed that after I posted this.

Thanks for your consideration. Can you call multiple `printer.add_object()` with the same parameter name? (ie `mmu_exit_leds`) ? I feel like the last is overwriting the first, and then the macro can only use the last one:

`SET_LED LED=mmu_{segment}_leds`

I'm willing to code this myself and submit a PR, but curious how you would tackle this. Create an indexed segment (mmu_exit0_leds, mmu_exit1_leds, etc)? Let me know what you think.

FWIW, I tried using the virtual_led plugin but it seems to be broken after a recent-ish klipper update. My thought was to add each exit led as it's own group and enumerate them individually.

JesNoFear, 2026-01-30:
this would be nice to see at some point.

i have nothing to add other than interest in this as well. I am mid-process of building two custom heated dryboxes with their own controllers, and would like (am currently printing) illuminated number badges for each of the spools, and to have the MMU status of the spool change the color according to its state, as well as selected spool be identified.

but its a want and not a need.

moggieuk, 2026-10-21:
Closing this, since v4 can now drive more than one LED per gate, and those LEDs can be on different strips.

In v4 every line of `exit_leds` (or `entry_leds`) in `[mmu_leds]` is added in order to one chain. That chain is then shared out evenly between the gates (or by `exit_led_counts` if you want an uneven split). So to mirror each gate on two strips, list them interleaved, one LED per line. For example, for a 3-gate unit:

```
exit_leds: neopixel:mmu_leds_1 (2)
           neopixel:mmu_leds_3 (16)
           neopixel:mmu_leds_1 (3)
           neopixel:mmu_leds_3 (17)
           neopixel:mmu_leds_1 (4)
           neopixel:mmu_leds_3 (18)
```

That's 6 LEDs over 3 gates, so each gate gets one rewinder LED and its matching skirt LED, and both show the same colour and effects. Just make sure the total is a multiple of the number of gates, and don't reuse the same physical LED in two segments.

If there's something v4 still doesn't do for you, please open a new feature request against v4. It's being actively worked on.

closed 2026-10-21 by moggieuk