moggieuk/Happy-Hare #624: extruder sync wierd behaviors

thisiscam, 2025-02-07:
Hi!

On recent version of HH, I have found various weird behaviors:
- At print end, after the gate is ejected, the extruder can still be synced to the toolhead. This can be problematic if there is any toolhead extruder movement, which will cause the current filament in the selected gate to protrude (or back off too much) from the gate. Also, there seems to be a major change in the sync behavior? --- a very long time ago, my implementation was so that sync would only be applied when filament is loaded, which now doesn't seem to be the case? 
- On print start, right after the initial toolchange, the extruder is somehow un-synced to the toolhead? See this piece of log:
```
         Gate Statistics:
         0:😃, 1:😎, 2:😊, 3:😎, 4:😎, 5:😎, 6:😃, 7:😃, 8:😃, 9:—, 10:😱, 11:😟, 12:😱, 13:😐
03:34:29   DEBUG: Continuing from printing state after toolchange
03:34:29   DEBUG: Sensors: {'mmu_gate': True, 'extruder': True, 'toolhead': True}
03:34:29     TRACE: Running macro: _MMU_CLEAR_POSITION
03:34:29     TRACE: Enabled runout detection
03:34:29       STEPPER: unsync()
03:34:29 Unsynced MMU from extruder
03:34:29     TRACE: Reset sync multiplier
03:34:29     TRACE: Setting gear motor rotation distance: 22.991720
03:34:29   DEBUG: Restoring MMU stepper_mmu_gear run current to 100% (1.25A)
03:34:30     TRACE: Processing idle_timeout 'ready' event
03:34:34     TRACE: Processing idle_timeout 'printing' event
03:58:50   DEBUG: Automatic clog detection: new min_headroom (< 5.0mm desired): 4.8mm
03:58:50   DEBUG: Automatic clog detection: new min_headroom (< 5.0mm desired): 4.2mm
03:58:50   DEBUG: Automatic clog detection: new min_headroom (< 5.0mm desired): 3.6mm
03:58:51   DEBUG: Automatic clog detection: new min_headroom (< 5.0mm desired): 3.0mm
```
This is causing me false positives of clog detection, since slack filament in the bowden path is moved first, the encoder is picking up less movement.
The issue is gone once I hit RESUME, which then the extruder syncs back correctly.

thisiscam, 2025-02-07:
Not sure if this is the cause of the second problem, since I have not dug into the HH source code for a long time:  https://github.com/moggieuk/Happy-Hare/blob/0588078d19dfe0e15d58599026c2299957f978a2/extras/mmu/mmu.py#L6200 
That comment says the next line should be outside of the `with` block --- but that doesn't match with the source code!

moggieuk, 2025-02-15:
What MMU are you using?

thisiscam, 2025-02-15:
> What MMU are you using?

A tradrack with binky with gate sensor.

thisiscam, 2025-02-15:
with the latest HH, the behavior is that the servo unsyncs, but the gear is still engaged. It's then  a clog caught by the encoder, and after resuming everything goes back to normal

moggieuk, 2025-02-15:
ok, looking...  the syncing behavior was changed to support the type-b designs like Box Turtle as well as to be deterministic at a number of time.  It is made more difficult with the in vs out of print use cases.  That said should be an easy fix...

moggieuk, 2025-02-15:
Ok, separating these two issues and tackling the end-of-print state first..

I assume that `MMU_PRINT_END` is being called.  I.e. you added to the end-gcode custom code?

If so, this should call "_on_print_end()" which ends with:
```
self.sync_gear_to_extruder(False, grip=True)
```
This determines what to do based on selector type and whether in print, but then:
```
        # Sync / Unsync
        new_sync_mode = MmuToolHead.GEAR_SYNCED_TO_EXTRUDER if sync else None
        if new_sync_mode != self.mmu_toolhead.sync_mode:
            self.movequeues_wait() # Safety but should not be required(?)
            self.mmu_toolhead.sync(new_sync_mode)
```
This should always unsync...  The only way I can see it being incorrect is if "filament_always_gripped" is set to true (which would be incorrect for Tradrack).   Hmmm...

I'm away from printer, but do you have ideas based on the above?

thisiscam, 2025-02-15:
> Ok, separating these two issues and tackling the end-of-print state first..
> 
> I assume that `MMU_PRINT_END` is being called. I.e. you added to the end-gcode custom code?
> 

Hmm, the doc says this wasn't necessary (only necessary for octoprint) so I don't have it in my slicer. Is that it? I will try after my print ends.

thisiscam, 2025-02-15:
Actually, that's still not so right. 
Right now I have 
```
MMU_END
END_PRINT
; MMU_PRINT_END # I don't have this right now
```

I think what you are saying is that the last  `MMU_PRINT_END` would do the unsync logic.
But, the filament would already be ejected by END_PRINT (my custom macro, for cleaning nozzle etc), but the gear is still synced to the extruder which seems incorrect?

moggieuk, 2025-02-15:
To clarify because I wasn't clear.
MMU_PRINT_END is the macro defined in Happy Hare that I expect to be run as a book-end.  However, it doesn't need to be called directly because it is called naturally at the end of a non-octoprint print from this line (2951):
```
self.reactor.register_callback(lambda pt: self._print_event("MMU_PRINT_END STATE=complete AUTOMATIC=1"))
```

So first it woud be look to see the `mmu.log` at the end of a print to see this is being called (it should be).  BTW all HH commands will be logged so expect to see:
`MMU_PRINT_END STATE=complete AUTOMATIC=1` in the log

moggieuk, 2025-02-15:
Looking at the second problem of unsync() ing on first toolchange...

moggieuk, 2025-02-15:
Update on the second problem... I think you found the exact cause although the comment isn't really complete in explanation.  I t should be indented so that it is outside of `_wrap_gear_synced_to_extruder`.  The reason is that the "wrapper" is to ensure consistent sync state both in an out of a print by restoring the entry state.  However I think that in on the initial tool change this might be causing the unsync(). 

```yml
            # Deliberately outside of _wrap_gear_synced_to_extruder() so there is no absolutely no delay after restoring position
            self._continue_after('toolchange', restore=restore)
        except MmuError as ee:
            self.handle_mmu_error(str(ee))
```

I'm going to look backward because the sync to extruder state should have already been established correctly before this method is called -- if it was it would have worked but the coding error you found would still be there.

thisiscam, 2025-02-15:
[mmu.log](https://github.com/user-attachments/files/18812813/mmu.3.log)

Here's my log prior to print start up to the detected clog, the log for this print starts at line 7208.

thisiscam, 2025-02-15:
Looking at my print_start macro, I'm not sure if this could be related: I have a complicated heat soak routine that puts the current print into paused state, and when the chamber is up to temperature, it resumes the print. To do that, I used the following "hack":
```
[gcode_macro _PAUSE_SKIP_MMU]
gcode:
    {% if printer.mmu is defined %}
        __PAUSE {rawparams} 
    {% else %}
        PAUSE {rawparams}
    {% endif %}

[gcode_macro _RESUME_SKIP_MMU]
gcode:
    {% if printer.mmu is defined %}
        __RESUME {rawparams}
    {% else %}
        RESUME {rawparams}
    {% endif %}
```
The idea is that the pause and resume should skip the MMU pause/resume logic (which I believe contains some sync related logic, some temperature manipulation due to "MMU_LOCK" and toolhead position saving etc.), because I still want to operate the machine normally from command line with the MMU being transparent. 

Could this have caused problem? I didn't have this problem back around v2.5. 

Nonetheless, here's my heat soak setup macros:

[print_setup.cfg.txt](https://github.com/user-attachments/files/18812826/print_setup.cfg.txt)

moggieuk, 2025-02-16:
Well, there was a missing initial syncing of extruder .. I don't know why nobody reported this sooner .. perhaps because synced would have been picked up after initial tool loading.  Anyway, I've pushed a fix for that.

On the syncing remaining on at print end .. I'm not seeing that.  In the log it is always unsynced as far as I can see.

Anyway, update to fix the print start sync and let me know how that works out..

thisiscam, 2025-02-16:
> Well, there was a missing initial syncing of extruder .. I don't know why nobody reported this sooner .. perhaps because synced would have been picked up after initial tool loading. Anyway, I've [...]

Thank you! Will try you fix later today.

For the syncing remaining on at print end, it wasn't in the above log because the print hadn't ended. Here's my full log

[mmu.log](https://github.com/user-attachments/files/18813631/mmu.4.log)

See around line 8465 --- the tool has unloaded at that point, but it still synced.

thisiscam, 2025-02-17:
@moggieuk I tried your fix and it seems to work as intended now! Thank you! The end of print sync problem is less of an issue for me, but would be great if fixable since the behavior can lead to hard-to-discover gate protrusion problems.

thisiscam, 2025-02-23:
@moggieuk I wanted to track down what is happening, and I don't think the fix was right?
The fix was:
https://github.com/moggieuk/Happy-Hare/blob/57f94e6bd23bb52a6753bd9b0fbdb6a4030ba299/extras/mmu/mmu.py#L3016

But, see this line above, the extruder should have been synced. 
https://github.com/moggieuk/Happy-Hare/blob/57f94e6bd23bb52a6753bd9b0fbdb6a4030ba299/extras/mmu/mmu.py#L3002

I believe the problem is more fundamental, and seems to be related to how the sync state is persist across prints.

thisiscam, 2025-02-23:
The problem is probably that this case is triggered: https://github.com/moggieuk/Happy-Hare/blob/57f94e6bd23bb52a6753bd9b0fbdb6a4030ba299/extras/mmu/mmu.py#L5418

closed 2025-04-08 by github-actions[bot]