moggieuk/Happy-Hare #633: Spoolman missing part of the filament consumption

igiannakas, 2025-02-10:
So while the spoolman spool assignment https://github.com/moggieuk/Happy-Hare/pull/619 fixed part of the issue, I am still missing around 40 grams of filament for the below print:

[image]

[image]

[image]

[image]

Started off with a full spool - 1354grams including desiccant etc. Ended up with 1195 grams, **so consumed 159 grams.**

Slicer had requested a total of 152 grams, with purge being 123 grams and model 11 grams, **so slicer estimate is close enough - 152 vs 159 grams.**

**However spoolman recorded ~119 grams consumed** so a delta of around 40 grams or so, which is larger than the model print, hence purging definately is counted but not all (?). This means approx 13 meters of filament are unaccounted for :)

Any ideas what may be going on here? Anyone else noticed this?

moggieuk, 2025-02-15:
Hmmm, I thought with PR #619 that the hole was completely closed now.  I.e. Happy Hare is always setting the active spool_id because any printing or purging.  It's possible that there is still an oversight, but are you sure you have the density set correctly in Spoolman?   I honestly haven't looked at how it does the calculation but I thought I remember it using `density x volume extruded`

igiannakas, 2025-02-15:
Yeah I thought so too - it helps but there is still filament missing :(

The density is set correctly and the same as the slicer - slicer, model and poop reconcile but spoolman doesn’t but only on multi color prints.

I’m thinking whether the unload process is the culprit - maybe the filament should be marked as inactive in spoolman before the unload process begins. I’m planning to run a test on it in a few days when the current print job batch is done and will see. 

Alternatively, we are always leaving a fragment of filament in the hotend that gets purged out but not accounted for in the current spool (the residual filament after cutting) but I’m not sure whether that’s enough to explain the delta. 

Will be testing more and let you know…

moggieuk, 2025-02-16:
Hmmm,  I'm sure that spoolman uses extruder movement to tally up consumption.  When loading or unloading a filament, the extruder is being controlled by my extra "toolhead" which would not be visible to spoolman.  The extruder only moves on the regular toolhead after loading is complete.  So I'm at a loss here.  Please continue to delve into this because I believe you I just don't know where to look (and I really have to get the Mainsail work complete..)

igiannakas, 2025-02-16:
Will do, it’s driving me up the wall too! First to replicate the result with another print, then to try and figure out why :)

igiannakas, 2025-03-18:
Keeping open to do the promised investigation

closed 2025-05-02 by github-actions[bot]