moggieuk/Happy-Hare #635: filament_motion_sensor definitions makes Klipper error out

HitLuca, 2025-02-12:
Defining an encoder sensor using the traditional `filament_motion_sensor` definition makes klipper break and throw the following error

```
Internal error during ready callback: 'EncoderSensor' object has no attribute '_extruder_pos_update_timer'
```

A sample encoder config is (for a btt sfs 2.0)

```
[filament_motion_sensor encoder_sensor]
switch_pin: ^PA0
detection_length: 7
extruder: extruder
pause_on_runout: False
runout_gcode:
  M118 Filament encoder runout
  PAUSE
insert_gcode:
  M118 Filament encoder inserted
```

ammmze, 2025-02-12:
I was able to reproduce this by adding a `filament_motion_sensor` AFTER all my mmu include directives. If I move the `filament_motion_sensor` to be BEFORE all of the mmu include directives, the printer starts up.

Based on the traceback and a peek at the code both in HH and Klipper, it looks like HH does some stuff at one of the klipper start up events that triggers some of the other events to fire off and ultimately causes an event to land in the `filament_motion_sensor` before the `filament_motion_sensor` has completed its set up.

HitLuca, 2025-02-13:
Can confirm this workaround works. A proper solution would be nice but unless people start doing complex multi-non-hh-encoder-sensor setups it should work fine

moggieuk, 2025-02-15:
Just to be clear on your use... firstly you can't define an MMU encoder with:
```
[filament_motion_sensor encoder_sensor]
```
and expect it to be used with Happy Hare.  You need to use this for that purpose:
```
[mmu_encoder encoder_sensor]
```

But I believe you are saying that adding an additional motion sensor with:
```
[filament_motion_sensor my_encoder_sensor]
```
doesn't work? (BTW I'd refrain from using the name "encoder_sensor" else it is confusing quickly with HH's default name).

Looking at the stack backtrace, this is actually a bug in Klipper's filament_motion_sensor (any actually my mmu_encoder which was originally based on that code).   The bug is that it doesn't define the variable `self._extruder_pos_update_timer` until `handle_ready()`.  This is a problem because other klipper modules can "activate" any other module (as is the case here) and thus lead to this problem.  I will fixed in `mmu_encoder` , although that will never see the problem because of the order of events but I can't fix klipper.   The only thing you can do is to define `filament_motion_sensor` later as @ammmze  has suggested.  Klipper does have some startup ordering corner-cases...

Note that I do recommmend adding Happy Hare includes early in the `printer.cfg` - in fact I always put them first.

HitLuca, 2025-02-17:
yes I was setting up an unrelated filament motion sensor not to be used by HH. The fix proposed works so I can say this issue is closed