moggieuk/Happy-Hare #637: Enhancement: Remove `servo_always_active` and `_active_down`

burkfers, 2025-02-14:
I've seen many reports of servos dying after a user enabled the setting to have the servo always active or active on down, but zero instances where enabling this would be helpful.

In a brief discussion on Discord, @norris108 suggested outright removing the option.

If the option is still useful in niche cases, the "CAUTION" option should be made more explicit to warn the user of imminent servo death along the lines of "CAUTION MAY DAMAGE SERVO".

moggieuk, 2025-02-15:
I can only so do much to protect the user — some servos now in use require this option.  It also is probably necessary for Tradrack.  I do default to off.

ningpj, 2025-02-15:
No, TR works fine without permanently enabling the servo.

thisiscam, 2025-02-18:
I am guilty for adding this option a long time ago. 
It was for a niche setup (three position servo TR), though I no longer user it. 
I think a big lousy comment here is good enough, while making sure it doesn't break existing users (whoever that might be!)

moggieuk, 2025-02-24:
The problem is that there are some folks who rely on this feature.  I forget the name but there is one servo that requires it (I believe a kit supplier is using it).  I'm happy to make wording stronger or ALL CAPITALS but I can't remove it without creating even more support problems.

The reality is that IF the servo can reach the desired position, keeping it on shouldn't be an issue, its just that so many don't calibrated the servo correctly

burkfers, 2025-02-24:
I've submitted PR #650 with an adjustment to make the wording more forceful. A one liner on boot-up sounds like a great idea, but is beyond my skills at this time.

burkfers, 2025-02-26:
I consider this request fulfilled (Thanks to everyone!) with the merging of #650, a harsher warning is sufficient.

closed 2025-02-26 by burkfers