moggieuk/Happy-Hare #645: Spoolman Extra Fields

jonwally, 2025-02-21:
Found that when Happy Hare spoolman_support is set to "push", any values stored in other "extra fields" for the spool are erased.

tlanfer, 2025-02-22:
I came across the same issue yesterday. Unfortunately, spoolman requires you to send the whole spool, including all extra fields, when updating a spool via PATCH. Personally, i think thats not good API design, and it would benefit other integrations as well to have the spoolman API support partial updates, for example via `Content-Type: application/json-patch+json`.

jonwally, 2025-02-22:
That seems like a very poor API. Leaves a lot of room for the integration to make mistakes.

tlanfer, 2025-02-23:
I've asked the spoolman dev for a change to make this easier for API consumers: https://github.com/Donkie/Spoolman/issues/638
Lets see how that goes.

tlanfer, 2025-02-26:
I have a PR prepared that fixes this: https://github.com/Donkie/Spoolman/pull/639
I'm running that version locally here already and its working great.

tlanfer, 2025-03-17:
My change made it into the v0.22.0 release of spoolman, so this should be fixed. i will try replacing my custom container later.

moggieuk, 2025-03-24:
Awesome.  Thanks for pushing to Spoolman!!   Going to close this issue now.

closed 2025-03-24 by moggieuk