moggieuk/Happy-Hare #653: mmu_leds.cfg enabled 'LEDHelper' object has no attribute 'get_led_count'

mochoandre, 2025-02-26:
Updated from 2.71 to 3.1.0 and updated klipper

When restarting I have error 
"Internal error during ready callback: 'LEDHelper' object has no attribute 'get_led_count'
Once the underlying issue is corrected, use the
"FIRMWARE_RESTART" command to reset the firmware, reload the
config, and restart the host software.
Printer is shutdown"

If I disable mmu_leds.cfg (changing file name extension) it starts normally. 
I am not using mmu happy hare Leds. Only my other neopixels (toolhead and case light) 

Just to let you know

moggieuk, 2025-02-26:
Hmmm.  Thanks. I'll take a look - I thought I tested without LEDs...

mochoandre, 2025-02-27:
```
# MMU OPTIONAL NEOPIXEL LED SUPPORT ------------------------------------------------------------------------------------
# ██╗     ███████╗██████╗ ███████╗
# ██║     ██╔════╝██╔══██╗██╔════╝
# ██║     █████╗  ██║  ██║███████╗
# ██║     ██╔══╝  ██║  ██║╚════██║
# ███████╗███████╗██████╔╝███████║
# ╚══════╝╚══════╝╚═════╝ ╚══════╝
# Define the led connection, type and length
#
# (comment out this section if you don't have leds or have them defined elsewhere)
[neopixel mmu_leds]
pin: mmu:MMU_NEOPIXEL
chain_count: 20			# Need number gates x1 or x2 + status leds
color_order: GRBW		# Set based on your particular neopixel specification (can be comma separated list)
```

```
[mmu_leds]
exit_leds:   neopixel:mmu_leds (1-9)
#entry_leds:  neopixel:mmu_leds (10-18)
#status_leds: neopixel:mmu_leds (19)
#logo_leds:   neopixel:mmu_leds (20)
frame_rate: 24
```

closed 2025-04-12 by github-actions[bot]