moggieuk/Happy-Hare #665: Upload of gcode files with metadata fails

dagolard, 2025-03-11:
Problem: I cannot upload Files with metadata.

made following test:
- fresh install of newest Klipper, mainsail and moonraker.
- uploaded a file from Cura

File uploaded perfect, size okay, thumbnail okay

- installed newest Happy Hare from Github
- uploaded the same file

File upload didn´t work. Size not okay, no file in directory, no thumbnail. Also moonraker reports a error.

[image]

Included the moonraker log...

[moonraker.log](https://github.com/user-attachments/files/19196711/moonraker.log)

ningpj, 2025-03-11:
Can you please upload and attach the unprocessed gcode from cura so the issue can be recreated

dagolard, 2025-03-11:
I do it later. But I tried Orca Slicer too - same result.

dagolard, 2025-03-11:
[image]

This appears after installing HH...

dagolard, 2025-03-11:
Update: tried with Orca, and it worked.
Cannot save the Cura ufp file local. There it is storing only the gcode part without thumbnail.

[Untitled.zip](https://github.com/user-attachments/files/19197359/Untitled.zip)

dagolard, 2025-03-11:
Strange effect: uploading the Cura gcode from PC via "Upload new gcode" button and THEN the ufp from Cura works. No idea why...

[image]

dagolard, 2025-03-11:
Have to go. CU tomorrow.

ningpj, 2025-03-12:
Try a few other print jobs from cura and see how you get on.  The cura file uploaded fine although it didnt contain any HH specific tokens the preprocesser might have potentially had issues with.  Check you have freespace on your SD card and there arent any odd IO related errors reported in ``dmesg`` or ``journalctl`` (syslog) on your pi

dagolard, 2025-03-12:
I tried various files. Have you tried in moonraker connection "ufp with thumbnail"?
Clean gcode upload works everytime.
Also the mmu_server error after HH install irritates. He appears with version 3.1. Would try a 3.0 installation (rollback) for testing it is with 3.1 only, but don't know how.

ningpj, 2025-03-12:
I dont know much about cura ufp files other than they are compressed archives and are not pure text files.  cura can imbed images in gcode just like orca/prusa/ss which is the format predominately used by HH users.  Need a ufp file to play around with to see how moonraker handles this file format.   WHat have you enabled in moonraker.conf specifically to support ufp file uploads?  e.g. 
  
[octoprint_compat]
enable_ufp: True

dagolard, 2025-03-12:
I don't know how to save it out of Cura, but here is the specification...

https://github.com/Ultimaker/libCharon/blob/main/Charon/filetypes/UltimakerFormatPackage.py

TWahl93, 2025-03-27:
I have the same problem with PrusaSlicer 2.9.1. The gcode has the typical "; processed by HappyHare" in the header. Before the update it worked fine

nasrabq, 2025-03-28:
I have the same issue. I have tried multiple slicers (Orca 2.2, Orca 2.3 and Prusa), and all have the same issues. before it was working but now no.

I was checking this issue in Mainsail Discord and found the below comments:
"_the mmu code combined with the addition of klipper estimator preprocessor to Moonraker and therefore a refactor of preprocessors_"

moggieuk, 2026-10-21:
Thanks for reporting this, and for noticing that it only worked if the plain `.gcode` had been uploaded first. That was the key clue!

The cause: when Happy Hare's Moonraker component is installed, it pre-processes each uploaded file before Moonraker reads its metadata. For a Cura `.ufp` file, though, Moonraker is the one that unpacks the `.gcode` from the `.ufp`, and Happy Hare was looking for the `.gcode` before it had been unpacked. It didn't find it and stopped the whole metadata step, so the upload failed.

This is now fixed in v4 (#1437) and will be in the next release. The `.ufp` is unpacked first (using Moonraker's own code, so thumbnails are handled the same way), and then Happy Hare pre-processes the extracted `.gcode` as usual. Plain `.gcode` uploads are unchanged.

The later comments about Prusa/Orca uploads timing out look like a separate issue. If anyone still sees that on v4, please open a new issue with your `moonraker.log`.

closed 2026-10-21 by moggieuk