moggieuk/Happy-Hare #668: 3dChameleon extruder gear direction not following the normal direction when preloading

TuanNguyenForben, 2025-03-12:
There is an issue with the 3dchameleon rotary selector type when preloading.

the gear motor direction when using preload won't change no matter the pin direction in the config and will just move to the opposite direction to the normal build.  To fix this I tried flipping the input and output side of the tool and works fine.

Below is my log when trying to preload with T0 and the output swapped.

[mmu(2).log](https://github.com/user-attachments/files/19218595/mmu.2.log)

moggieuk, 2025-03-17:
I see the oversight in the code.. working on solution

moggieuk, 2025-03-17:
I wonder if this is a h/w setup mismatch....   you have the gate sensor set to be "mmu_gate".  This is a switch that should be in the filament path AFTER the MMU gears.  Any sensor before that would be a "mmu_pre_gate" sensor and would be per-gate/lane.

Can you describe the sensors you have connected to you 3dchamelon. I.e. imagine inserting a filament, describe the path to filament takes with all sensors you have fitted...

TuanNguyenForben, 2025-03-18:
@moggieuk i think it is the correct configuration from the documentation, thee reason why you see errors in the log is probably because I have not added a mmu_gate configured but here is my set up right now with 4 post gear sensor 

[image]

[image]

moggieuk, 2025-04-29:
Hi, I've really sorry -- I forgot about you!

How is this going?    The 3DChameleon is a little different from other MMU's in that it has these two extra parameters:
```
cad_gate_directions: 1, 1, 0, 0
cad_release_gates: 2, 3, 0, 1
```

I've shown with the built in defaults.  The way this works is that if gates 0 or 1 are selected then the rotation of the stepper is reversed.  If gates 3 or 4 are selected the the stepper is in the normal forward direction.

Could it be that your gates are combined in a different order meaning the defaults above aren't correct?

moggieuk, 2026-10-21:
I'm closing this as part of a tidy-up of old v3 issues. Happy Hare v4 is a massive architectural change from v3, so it isn't possible to address these on v3 any more.

The 3DChameleon selector was reworked in v4 and now sets the gear direction for the selected gate each time it grips or releases.

If this still exists in Happy Hare v4 (v4.1 is the current release), **please resubmit** it as a new issue (with your `mmu.log` if relevant) and it will be reviewed.

closed 2026-10-21 by moggieuk