moggieuk/Happy-Hare #670: Filament change temperature adjust-Different material types-Timing of temp change in toolchange process.

53Aries, 2025-03-14:
When using different materials during a print the point at which the hotend temperature is adjusted is slightly out of place. 
The current process is:
Toolchange requested
Change hotend temperature to next 
Ramming/tip forming
Unload
Load

It would be better to have the temperature not change until after unloading:
Toolchange requested
Ramming/tip forming
Unload
Change temperature
Load

[image]

[mmu (7).log](https://github.com/user-attachments/files/19255018/mmu.7.log)

53Aries, 2025-03-14:
Huh. this is weird. Looking at the mmu_form_tip.cfg I see that with toolchange_temp set to 0 in gcode_macro _MMU_FORM_TIP_VARS(which mine is set to 0) it shouldn't be changing temp until M104 S{next_temp} is called which IS after ramming/tip forming....SO I have no idea why mine is changing temp before ramming/tip forming.

moggieuk, 2025-03-24:
Temp turn on TRACE file logging with:
```
MMU_TEST_CONFIG log_file_level=3
```
Then capture mmu.log.  You should see a line like this:
```
_ensure_safe_extruder_temperature: current_temp=%s, paused_extruder_temp=%s, current_target_temp=%s, klipper_minimum_temp=%s, default_extruder_temp=%s, source=%s
```
with info filled out. This will give a clue.   But when printing the logic will default to the current target temp which will be set by the slicer..

DrEvil23, 2025-11-15:
Hi everyone,

I have the same problem : 

16:29:11   DEBUG: Continuing from printing state after toolchange
16:29:11     TRACE: Running macro: _MMU_RESTORE_POSITION
16:29:11 Restoring toolhead position to last pos: (x:132.8, y:58.1, z:32.3)
16:29:11   DEBUG: Clearing saved toolhead position
16:29:11   DEBUG: Ensuring correct gcode state and position (X:132.8 Y:58.1 Z:32.3 E:374645.7) after toolchange
16:29:11     TRACE: Enabled runout detection
16:29:17     TRACE: MmuSyncFeedbackManager: Sync direction changed from static to extrude
16:29:43   DEBUG: Setting rising toolchange floor to 32.30928
16:29:43 > MMU_CHANGE_TOOL TOOL=0
16:29:43     TRACE: Disabled runout detection
16:29:43   DEBUG: Tool change initiated with standalone MMU tip forming and slicer purging
16:29:43 Tool change requested, from T3 to T0
16:29:43   DEBUG: Saving toolhead gcode state and position (X:101.3 Y:39.1 Z:32.3 E:-1.0) for toolchange
16:29:43     TRACE: Running macro: _MMU_SAVE_POSITION
16:29:43 Saving toolhead position (x:101.3, y:39.1, z:32.3)
16:29:43     TRACE: Running macro: _MMU_PARK
16:29:43 Parking toolhead at (x:200.0, y:n/a, z:32.3) for toolchange operation
16:29:43     TRACE: Running macro: SET_GCODE_VARIABLE MACRO=_MMU_PARK VARIABLE=next_pos VALUE=False
16:29:43   DEBUG: Unloading tool T3
16:29:46     TRACE: Running macro: _MMU_ACTION_CHANGED ACTION='Unloading' OLD_ACTION='Idle'
16:29:46     TRACE: Running macro: _MMU_PRE_UNLOAD
16:29:46 Unloading filament...
16:29:46       STEPPER: [mmu_gate sensor is TRIGGERED]
16:29:46 [T3] <<< (*) << En < [Ex <<< Nz] LOADED 0.0mm (e:0.0mm)
16:29:47     TRACE: Running macro: _MMU_ACTION_CHANGED ACTION='Cutting Tip' OLD_ACTION='Unloading'
16:29:47     TRACE: _ensure_safe_extruder_temperature: current_temp=221.23, paused_extruder_temp=None, current_target_temp=255.0, klipper_minimum_temp=170.0, gate_temp=200, default_extruder_temp=200.0, source=auto
16:29:47     TRACE: Running macro: _MMU_ACTION_CHANGED ACTION='Heating' OLD_ACTION='Cutting Tip'
16:29:47 Waiting for extruder to reach target (slicer) temperature: 255.0°C
16:30:01     TRACE: Running macro: _MMU_ACTION_CHANGED ACTION='Cutting Tip' OLD_ACTION='Heating'
16:30:01   DEBUG: Setting pressure advance for tip forming: 0.0000
16:30:01 Forming tip...
16:30:01     TRACE: Running macro: _MMU_CUT_TIP 
16:30:01 Retracting filament 33.0mm prior to cut
16:30:02 > MMU_STATS COUNTER=cutter_blade INCR=1
16:30:02 Pushing filament fragment back 18.0mm after cut
16:30:06   DEBUG: Restoring pressure advance: 0.0440
16:30:06     TRACE: After tip cutting, park_pos reported as: 51.0mm with calculated 7.0mm filament remaining in extruder (extruder moved: 34.0mm, encoder measured 38.7mm)
16:30:06       STEPPER: [mmu_gate sensor is TRIGGERED]
16:30:06 [T3] <<< (*) << En < [Ex <<. Nz] -51.0mm (e:51.1mm)
16:30:06     TRACE: Running macro: _MMU_ACTION_CHANGED ACTION='Unloading' OLD_ACTION='Cutting Tip'
16:30:06     TRACE: Running macro: _MMU_POST_FORM_TIP
16:30:06     TRACE: Running macro: _MMU_ACTION_CHANGED ACTION='Exiting Ext' OLD_ACTION='Unloading'
16:30:06   DEBUG: Extracting filament from extruder
16:30:06     TRACE: _ensure_safe_extruder_temperature: current_temp=257.26, paused_extruder_temp=None, current_target_temp=255.0, klipper_minimum_temp=170.0, gate_temp=200, default_extruder_temp=200.0, source=auto
16:30:06       STEPPER: [extruder sensor is TRIGGERED]
16:30:06   DEBUG: Reverse homing up to 140.0mm off extruder sensor (synced) to exit extruder
16:30:06       STEPPER: resync(extruder+gear --> gear+extruder)
16:30:06       STEPPER: _quiesce_align_get_tcut(full=True, wait=True) Elapsed:0.359609
16:30:06       STEPPER: unsync()
16:30:06   DEBUG: MmuSyncFeedbackManager: Unsynced MMU from extruder

As you can see, the temperature change is done before the cutting action.

Thank you,
Benoit

DrEvil23, 2025-11-16:
And it's still heating up too soon, without waiting this time...  It's like on every action it confirms if the extruder is hot enough, but it shouldn,t heat up until loading the new filament into the toolhead...  There is no log entry, but you can see the nozzle temperature getting up when unloading the previous filament...

moggieuk, 2026-10-21:
Thanks all, and thanks @DrEvil23 for the log, which shows what's going on. When `MMU_CHANGE_TOOL TOOL=0` starts, the extruder target is already 255°C. Your slicer (or your tool-change G-code) set the new tool's temperature before it sent `T0`. While printing, Happy Hare deliberately follows whatever temperature the slicer has set, so it waits for 255°C before cutting and unloading the old filament. Commenting out those calls in `mmu.py` only moves the problem, because the slicer has already changed the target.

What helps today:
- In your slicer's change-filament / tool-change G-code, make sure the new temperature is set after the `T` command (or keep the old filament's temperature until then), not before it.
- To heat for the new filament just before it loads, set `variable_user_pre_load_extension` in `mmu_macro_vars.cfg` to a small macro that sets the temperature. #393 has a worked example of this approach.
- If you use HH's standalone tip forming (not the cutter), `variable_toolchange_temp` in `_MMU_FORM_TIP_VARS` controls the temperature during tip forming.

Having HH itself apply the next tool's temperature at the right point in the toolchange is the same request as #393, so I'm closing this as a duplicate and tracking it there. If there's something else v4 doesn't do for you, please open a new feature request against v4. It's being actively worked on.

closed 2026-10-21 by moggieuk