moggieuk/Happy-Hare #673: Filament tension sensor acts like runout since update?

Christanoid, 2025-03-15:
Hello there,
I updated my voron recently and noticed that ever since I did all updates (did klipper and happyHare update, therefore I sadly cannot say which update caused this behaviour) my filament tension sensor from my belay is considered a filament runout sensor and triggers runouts constantly. 
As a workaround it luckily also appeared in my Miscellaneous section of Mainsail with the ability to disable it:

[image]

I don't know what configured it to also be seen as a Runout or how to get rid of it.
My endstop is configured unter the [mmu_sensors] section as the sync_feedback_tension_pin, doing a quick grep through my configuration folder I see noone else using that pin for any other endstop.

This bug pretty much makes the Belay useless, as I also don't see the small pod moving anymore, it just slowly moves towards the endstop once and then stays on that side. Any help would be appreciated, if you need any configuration files feel free to ask.

[mmu (2).log](https://github.com/user-attachments/files/19263295/mmu.2.log)

moggieuk, 2025-03-17:
Hi, I don't think that the tension sensor is firing a runout -- it isn't set up to do so, but I see that the pulling the slack in Belay IS causing a runout reported by the encoder:
```
16:31:35   DEBUG: Continuing from printing state after toolchange
16:31:35   DEBUG: Clearing saved toolhead position
16:31:36   DEBUG: Setting rising toolchange floor to 21.0
16:32:41   DEBUG: Automatic clog detection: new min_headroom (< 5.0mm desired): 3.7mm
16:34:31   DEBUG: Automatic clog detection: new min_headroom (< 5.0mm desired): 2.5mm
16:34:31   DEBUG: Automatic clog detection: new min_headroom (< 5.0mm desired): 0.6mm
16:34:32   DEBUG: Automatic clog detection: new min_headroom (< 5.0mm desired): -0.8mm
16:34:32   DEBUG: Automatic clog detection: new min_headroom (< 5.0mm desired): -2.4mm
16:34:32 > __MMU_ENCODER_RUNOUT
```

The "buffer" that belay or similar devices introduces into the bowden can be problematic with encoder based logic which is using a mismatch of extruder stepper movement and that measured by the encoder...

On the error reported, the "clog" is determined to be a "runout" because the encoder is not reporting movement when the filament is "buzzed" (moved back and forth).  This is unusual and suggests an encoder malfunction.   Something to look at.

In summary, the encoder clog/runout could have been caused by encoder malfunction but let's assume it isn't -- you can do two things to help:
1. If you are not synchronizing gear/extruder, then `toolhead_post_load_tighten` can be used to tighten the filament before resuming. It's a % of clog detection length and gets rid of slack that can lead to false clog detection.
2. Increase the "clog detection length" manually and/or turn off the automatic adjustment mode (`enable_clog_detection`).

Christanoid, 2025-03-18:
Ah I see, good to know.
On another note, just for understanding / debugging purposes, what does the toggle from my Screenshot do? Does it simply prevent the endstop from reporting triggered or is there a deeper logic there? I am still lightly confused because turning that toggle off prevents my printer from stopping entirely.
I'll have to do more calibration when I find the time, but for now I think the error lies in my rotation_distance for the actual hotend extruder and the synced mmu extruder. There appears to be a lot of variance in their actual extrusion since I initially needed quite large tension correction values (0.75 and 1.15 I think it was) for the binky capsule to move at all. I will try to recalibrate them again with the usual "extrude 100mm and measure how much was extruded" method.

But since it most likely is mechanical issues on my side i'll close this issue already.

closed 2025-03-18 by Christanoid

AlexTOOT, 2025-04-19:
>  since I initially needed quite large tension correction values (0.75 and 1.15 I think it was) for the binky capsule to move at all. 

Hi. how is everything going? I also have the problem. I calibrated both extruder and mmu gear stepper, when I test a 400mm move, extruder goes exactly 400 and  mmu gear goes about 399 or so, this is extremely accurate as I take it.

but, in print, they differs alot and I also have to set the ratio to extremely big values (I use 2.0 and 0.75), but i still get filament tension issue.